betrusted-io/xous-core #4: Why not SeL4?

cwebber, 2020-11-02:
Just curious, and it's ok to close this after explaining... I wasn't sure where else to open it though and I suspect I won't be the only one to wonder.

Why a special micokernel instead of using the already verified SeL4 effort?  Some special performance consideration, or?

(Also, are you planning on providing similar capability-paradigm security approaches?)

xobs, 2020-11-03:
To answer the Capability question: yes. There will be a `Pledge` syscall that will remove capabilities. That way a program can `Pledge` that it won't use the `Connect` syscall, which will prevent it from making any further calls, and will incur a crash if it does.

As for the reason of going with a special microkernel, I wanted something that directly implemented Rust-style semantics. Passing memory is somewhat of an afterthought with SeL4, with it preferring to pass values in registers. This is similar to a `Scalar` or a `BlockingScalar` call in Xous.

Instead, I wanted to try something closer to the Qnx approach. The `Move`, `Borrow`, and `MutableBorrow` messages map nicely onto Rust semantics, and should be immediately understood by most Rust programmers.

I also wanted to double-down on Rust. The kernel is Rust, userspace is Rust, so everything is Rust. It forces us to understand how all the parts will work together, and forces us to really consider things such as the kernel<->userspace barrier and how page moving works.

Eventually we'll provide a C library for Xous, but for now it's Rust all the way.

Finally, being able to do `cargo xtask run` is a nice bonus. By allowing the kernel to be hosted on another operating system, we allow using normal tools such as gdb, lldb, or visual sudio for debugging and profiling.

cwebber, 2020-11-24:
> To answer the Capability question: yes. There will be a Pledge syscall that will remove capabilities. That way a program can Pledge that it won't use the Connect syscall, which will prevent it from making any further calls, and will incur a crash if it does.

Hi... hm, that doesn't sound like an object capability system to me.  Note that there's a distinction between what are called "capabilities" in the object capability world vs for instance in the linux kernel... two very different things.  Here I mean reference-based authority.  I can explain more if that will help.

Thank you for answering though!

cwebber, 2020-11-24:
To understand the difference between object capabiltiies and posix capabilities (or "crapabilities"), read [this post by Kenton Varda](https://lwn.net/Articles/641534/).

bunnie, 2021-01-15:
Since this issue is being referred to by others, just adding my 2c as posted in an IRC chat:

`I can't speak for xobs on his motivations but I'm personally very on board with Xous for a number of reasons. Maybe the most honest and important one is that I just want to understand every nook and cranny of the OS and implementing one from scratch is a fun way to do that. Much more fun that just reading someone else's code and nodding your head at a bunch of decisions you don't really understand. But an important corollary of that is this dovetails nicely into the "trust because it's simple enough to actually understand" requirement, under the logic that I'm not a very good Rust programmer and if it's simple enough for me to understand the entire body of code then probably experienced Rust programmers can realistically audit the entire codebase in a non-annoying amount of time.

The "supply chain" for the core kernel is a very limited number of external packages, and I'd like to keep it at that -- every external dependency contains the potential for someone to inject lines of code that are in our blind spots.

Another facet is that it's an opportunity to rethink the traditional boundaries between processes, security, and the kernel. So a pipe dream of mine is to redo the POSIX file system to incorporate plausible deniability from the bottom-up, and to build a UI manager that contains notions of what is trustable content at a primitive level, under the theory that the UI manager can then render untrusted content in a way that's visually very different from trusted content so as to make it harder to spearphish someone into typing passwords/PINs into things that look substantially similar to secure fields but are actually just bitmaps crafted to look like trustable fields.

That being said, I think Sel4 is pretty cool and I really hope someone ports it to Precursor. It covers a different set of use cases and priorities and might be more suitable for many target audiences.`

bunnie, 2021-01-15:
Also re: capabilities, sorry for the confusion on terminology. We are aware that the capabilities of processes are different from pointer capabilities, but there is some name space collision which probably we should rectify. 

Regarding pointer capabilities, I am personally a big fan of them, but I prefer them to be implemented in hardware (I guess that's my bias as the hardware guy). We have looked at architectures like CHERI; there's some subtleties like how do you create capabilities for I/O blocks and what not that we've talked about ways to solve, but either way it would require non-trivial modifications to the internal bus interconnect to handle the capability bit. The other issue is that for CHERI they targeted C. They do use llvm so there's talk about making a Rust target but this is one of those "hm maybe we have to raise money to do that" sort of conversations. 

Rust in and of itself of course has a pretty nifty memory safety model, and we combine that with traditional virtual memory-based page isolation to bootstrap us into what I think is a medium-happy place for security. If we can find a way to layer pointer capabilities into this that would be cool but personally I'm very motivated to use this experience to learn more Rust and explore alternative paradigms for OS construction, which I hope someday could include capabilities but it would be...maybe a bit down the road once the compiler and hardware infrastructure for incorporating capabilities is a bit more mature.

zarutian, 2021-01-26:
> Since this issue is being referred to by others, just adding my 2c as posted in an IRC chat:

Re POSIX and capabilities see [CloudABI](https://docs.rs/cloudabi/0.1.0/cloudabi/) for a subset of POSIX that is compatible.

bunnie, 2022-05-23:
closing this so we can clean up the issue stack

closed 2022-05-23 by bunnie