betrusted-io/xous-core #7: add xtask task to update submodules

bunnie, 2020-11-25:
now that we have submodules enhance xtask to update the submodules

bunnie, 2020-11-25:
actually, a better way may be to have the Cargo.toml refer directly to a git repository, from @xobs :

```
With the com_rs = { ... } declaration above, it will clone the git repository from that URL, but it will always build at the specified git revision. When you're happy with a change, you push your com_rs repo and then update the Cargo.toml file under services/com/. Push as much as you like, as long as you don't force-push and remove that git revision, older versions will continue to work.
```

development would happen with this procedure:

```
   1. Clone com_rs somewhere
   2. Modify Cargo.toml to point to your local com_rs using com_rs { path = ... }
   3. Make & test changes
   4. Push changes
   5. Update Cargo.toml to use the new git rev
```

xobs, 2020-11-25:
The `declaration above` is:

```toml
com_rs = { git = "https://github.com/betrusted-io/com_rs", rev = "9a91a81f482ea8eba2f2de5d6c20b9bd399f39a9" }
```

bunnie, 2020-11-27:
I think this is resolved. @xobs please re-open it if you think it needs more work.

closed 2020-11-27 by bunnie