betrusted-io/xous-core #8: graphics-server: what API to use to specify Fonts

bunnie, 2020-11-27:
The currentl graphics-server uses `GlyphSet` to specify a font. 

I noticed in blitstr that instead, a `Style` API was introduced to specify a font during `PaintStr`.

Can I get some guidance as to what is the intended method to specify a typeface for the `blitstr` library, and how it might work?

samblenny, 2020-11-27:
`GlyphSet` is an implementation detail of the font module that has to do with abstracting access to the static arrays of glyph blit pattern data. It will probably change, and I don't recommend relying on it.

`Style` is part of the public API in `lib.rs` that I hope to keep relatively stable going forward. `Style` it's probably best understood as an abstraction for selecting between strategies for resolving ambiguity when looking up glyphs that are present in more than one GlyphSet.

In this context, "ambiguity" means more than one GlyphSet provides a glyph for the exact same grapheme cluster. For example, you could have a Latin Capital Letter A in Bold, Small, and Regular that all correspond to the grapheme cluster `41`. Requesting `Style::Bold` would resolve the ambiguity.

The same sort of ambiguity could come up for 骨 (grapheme cluster `9AA8`) or 返 (grapheme cluster `8FD4`) which have different region specific glyph variants. In the future, extending the `Style` enum to include options for `SimplifiedChinese`, `Japanese`, `Korean`, etc. would be a possible way to resolve that sort of ambiguity.
 
On the other hand, the Emoji glyph set is not really ambiguous in the same way. Blitstr only has one emoji typeface, and that probably won't change. There is some overlap in grapheme clusters between, for example, # (grapheme cluster `23`) and #️⃣  (grapheme cluster `23-FE0F-20E3`) where the first Unicode Scalar (aka codepoint) in the cluster is the same, `23`.

But, the Bold, Regular, and Small glyph sets only have a glyph for `23`, and the Emoji glyph set only has a glyph for `23-FE0F-20E3`. The grapheme cluster segmentation parser will try to match the longer grapheme cluster first (Emoji), and if that fails, it will then consider Latin glyph variants. Requesting a Latin `Style` variant when painting a string for the grapheme cluster `23` would be important for selecting which glyph you got. But, for `23-FE0F-20E3`, while you would need to specify a `Style` to make `paint_str()` happy, doing so would not affect which glyph would be used (because that cluster is unambiguous).

bunnie, 2020-11-28:
I understand now where I misunderstood the API. I was thinking of it from a standpoint of a Font-driven engine, where the font file used to render the glyphs is the invariant requested by the user. 

I think I now understand it as, `blitstr` will take a string of unicode characters, and it does its best effort to search the space of available blit patterns to render something on the screen. So if I pass regular text mixed with Emoji, it's intelligently auto-switching between the various Latin bitmaps and Emoji based upon the unicode region that's been specified. This is different from say, some Emoji fonts where they may not code Latin characters and if you pass the letter `a` to be rendered it could just give you the "character not available" symbol. 

In this case, yes, Style makes the most sense. I need to do do another pass of refactoring the message passing API then to reflect this. I'll also absorb your 0.3 changes into the message passing API and do a fairly major overhaul, but hopefully this fixes all the major inelegances with the `blitstr` interface into Xous so that I can start writing more substantial UI code into Xous.

closed 2020-11-28 by bunnie