betrusted-io/xous-core #12: Messages arrive out of order

bunnie, 2020-11-30:
It seems the order of messages is not guaranteed currently by Xous. This causes problems for the graphics server, because draw operations are being re-ordered, and this causes glitches and artifacts. 

Some form of message ordering is needed. At the very least, it would be good to have a guarantee that all messages from the same client arrive in order to a server. Relative ordering of messages from different clients does not need to be guaranteed.

xobs, 2020-12-04:
I added a test that checks for this, but I haven't been able to reproduce it in a testcase yet.

bunnie, 2020-12-04:
I could also believe it's the case that this isn't the actual root cause. Could be just a a screen update call that's copying frame buffer to screen but the frame buffer is still being drawn over.

I'll do another audit of the graphics-server to see if I can find something like that.

bunnie, 2021-01-08:
I suspect this is not an actual bug. There are still glitches on the drawing and update but I am more inclined to believe this is due to a race condition somewhere. I'm going to close this issue, and open a new one if I can find a commit that reproduces the issue solidly.

closed 2021-01-08 by bunnie