betrusted-io/xous-core #13: Refactor `create_server()` API

bunnie, 2020-12-03:
In order to assist with the implementation of the `xous-names` name resolution protocol, the API to create servers needs to be refactored.

* The "standard" call `create_server()` will have its argument removed, and it will just return a random, 128-bit number
* There will be a "kernel-only" call `create_server_named(name)` (or something like that) so that "well known" services like the name resolver and logging server can be established. This call is `pledged`-out by most processes before they are even started.

xobs, 2020-12-04:
I updated the issue to add checkboxes.

xobs, 2020-12-04:
This was completed in 168ffe004ee2514bb87cbcd45d9ea718caf85ea0

closed 2020-12-04 by xobs