betrusted-io/xous-core #14: New calls needed for `xous-names` to broker connections

bunnie, 2020-12-03:
`xous-names` is the de-facto connection brokerer. It keeps SIDs private while facilitating connections. 

In order to do this, two new calls are needed:

* `MessageSender.pid()` is needed by `xous-names` to extract the PID of the requesting process
* `ConnectForProcess(PID,SID)` is needed by `xous-names` to do the connection.

`ConnectForProcess()` should be exclusive to `xous-names` and this is done using the `pledge()` mechanism.

xobs, 2020-12-04:
You should now be able to call `sender.pid()`. It has been lightly checked to see if it works, but a thorough test has not yet been conducted.

xobs, 2020-12-04:
Both of these have been implemented.

bunnie, 2021-01-08:
These have been tested, and they work. closing.

closed 2021-01-08 by bunnie