betrusted-io/xous-core #21: Idioms for moving Structs using Xous Messages

bunnie, 2020-12-14:
In order to implement name lookup registrations (and probably many other things), it'd be nice to have an idiomatic example of how to pack a struct like this into a Xous message, and have its life cycle properly managed:

```
pub struct Registration {
    sid: SID,
    name: [u8; 64],
    name_len: usize,
    success: bool,
    _padding: [u8; 4096-(core::mem::size_of::<SID>() + 64 + core::mem::size_of::<usize>() + core::mem::size_of::<bool>],
}
```

Figuring out how to do this is a notch too hard for my Rust capabilities, so I'm hoping to enlist @xobs to help ponder this and set me in the right direction.

bunnie, 2020-12-14:
We also need to be able to, for example, return a Lookup struct that can optionally kick off an Authentication request:

```
pub struct Lookup {
    sid: SID,
    name: [u8; 64],
    name_len: usize,
    success: bool,
    autheticate: bool,
    challenge: [u32; 4],
    _padding: [u8; 4096-(core::mem::size_of::<SID>() + 64 + core::mem::size_of::<usize>() + core::mem::size_of::<bool>],
}
```

bunnie, 2021-01-02:
Do you have any docs on the lifecycle of a string? i noticed that strings are sort of like &[u8]s that can move around, but i was unable to implement copying of a &[u8] manually based on the string template.

xobs, 2021-01-02:
The way I would do this is by writing:

```rust
#[repr(C)]
pub struct Registration {
    sid: SID,
    name: [u8; 64],
    name_len: usize,
    success: bool,
}

#[repr(C)]
pub struct Lookup {
    sid: SID,
    name: [u8; 64],
    name_len: usize,
    success: bool,
    autheticate: bool,
    challenge: [u32; 4],
}

#[repr(C)]
union NameserverRecord {
    registration: Registration,
    lookup: Lookup,
}
```

Here, `[u8; 64]` is an array of 64 u8 values. This is different from a `&[u8]`, which is a slice, i.e. fat pointer to a series of u8 values. It's the difference between `char string[64]` and `char *string`.

The fat pointer can move around, but the thing it's pointing to is less likely to do so. In fact, I can't find any circumstances where the pointee would move, mostly it's just the pointer moving around.

bunnie, 2021-01-07:
`xous::ipc` crate solves this issue. marking as closed.

closed 2021-01-07 by bunnie