betrusted-io/xous-core #57: Reproducible builds and info leakage about the builder in binaries

bunnie, 2021-04-21:
Our build system is Rust's default build system, which by default includes things like the full path of the code (including the username) inside the binaries (for panic messages, strings, etc.). 

This makes the builds un-reproducible, and would also lead to compromise of users' identities if they are building their own binaries and loading it onto their devices. 

There are some things that could be done to improve this:

*    https://github.com/rust-lang/rust/issues/40374 (How can I stop rustc including system specific information such as absolute file paths of the source it's compiled from in the binaries it generates?)
*    https://github.com/rust-lang/rust/issues/40552 (Enable --remap-path-prefix for absolute paths by default)
*    https://github.com/rust-lang/rust/issues/60105 (no way to not ship panic strings inside the compiled binary)
*    https://github.com/rust-lang/cargo/issues/5505 (Reproducible builds: Automatically remap $CARGO_HOME and $PWD)
*    https://github.com/rust-lang/rust/issues/75799 (Cargo registry paths are leaked in compiled binary)

We'd love some help from someone who is more skilled in reproducible builds than us to rework our methodology to address this problem.

eupn, 2022-08-14:
[Nix](https://github.com/NixOS/nix) with [cargo2nix](https://github.com/cargo2nix/cargo2nix) may help.

jeandudey, 2023-03-20:
Also adding to this, GNU Guix is another option for reproducibility.