betrusted-io/xous-core #64: unable to lend buffers larger than 1 page

bunnie, 2021-05-04:
Something is preventing Xous from lending buffers larger than 1 page.

Here is a test case: https://github.com/betrusted-io/xous-core/commit/ea06df6879d740e33a1f30331c0c54aa25f41791

This is the KP:

```
PANIC in PID 8: panicked at 'unable to lend 60023000 in pid 8 to 40000000 in pid 7: ShareViolation', src/services.rs:1196:21
SYSTEM HALT: CPU Exception on PID 8: Instruction page fault of 0xffd0ac70 at 0xffd0ac72
Current thread 1:
PC:ffd0ac72   SP:fffefc70   RA:ffd0ac72
GP:00000000   TP:00000000
T0:00000063   T1:ffd0bdae   T2:ff400000
T3:40000001   T4:ff800000   T5:ff801000   T6:00000003
S0:0000001a   S1:4005a000   S2:81c40f18   S3:00000001
S4:fffff000   S5:82040f0d   S6:00000007   S7:ff400000
S8:00000ffc   S9:00000800  S10:00000800  S11:ff800000
A0:00000002   A1:ffd0dcff   A2:ffcf0000   A3:0000000a
A4:00000000   A5:ffd016f6   A6:fffefa8c   A7:ffc00f88
Memory Maps for PID 8:
       0 Superpage for 00000000 @ 40f03000 (flags: VALID)
          16 00010000 -> 40f93000 (flags: VALID | R | USER | A | D)
          17 00011000 -> 40f92000 (flags: VALID | R | USER | A | D)
          18 00012000 -> 40f91000 (flags: VALID | R | USER | A | D)
          19 00013000 -> 40f90000 (flags: VALID | R | X | USER | A | D)
          20 00014000 -> 40f8f000 (flags: VALID | R | X | USER | A | D)
          21 00015000 -> 40f8e000 (flags: VALID | R | X | USER | A | D)
          22 00016000 -> 40f8d000 (flags: VALID | R | X | USER | A | D)
          23 00017000 -> 40f8c000 (flags: VALID | R | X | USER | A | D)
          24 00018000 -> 40f8b000 (flags: VALID | R | X | USER | A | D)
          25 00019000 -> 40f8a000 (flags: VALID | R | X | USER | A | D)
          26 0001a000 -> 40f89000 (flags: VALID | R | X | USER | A | D)
          27 0001b000 -> 40f88000 (flags: VALID | R | X | USER | A | D)
          28 0001c000 -> 40f87000 (flags: VALID | R | X | USER | A | D)
          29 0001d000 -> 40f86000 (flags: VALID | R | W | USER | A | D)
     256 Superpage for 40000000 @ 40061000 (flags: VALID)
     384 Superpage for 60000000 @ 40031000 (flags: VALID)
           0 60000000 -> 40034000 (flags: VALID | R | W | USER | A | D)
           2 60002000 -> 40045000 (flags: VALID | R | W | A | D)
           3 60003000 -> 00000000 (flags: R | W)
           4 60004000 -> 00000000 (flags: R | W)
           5 60005000 -> 00000000 (flags: R | W)
           6 60006000 -> 00000000 (flags: R | W)
           7 60007000 -> 00000000 (flags: R | W)
           8 60008000 -> 00000000 (flags: R | W)
           9 60009000 -> 00000000 (flags: R | W)
          10 6000a000 -> 00000000 (flags: R | W)
          11 6000b000 -> 00000000 (flags: R | W)
          12 6000c000 -> 00000000 (flags: R | W)
          13 6000d000 -> 00000000 (flags: R | W)
          14 6000e000 -> 00000000 (flags: R | W)
          15 6000f000 -> 00000000 (flags: R | W)
          16 60010000 -> 00000000 (flags: R | W)
          17 60011000 -> 00000000 (flags: R | W)
          18 60012000 -> 00000000 (flags: R | W)
          19 60013000 -> 00000000 (flags: R | W)
          20 60014000 -> 00000000 (flags: R | W)
          21 60015000 -> 00000000 (flags: R | W)
          22 60016000 -> 00000000 (flags: R | W)
          23 60017000 -> 00000000 (flags: R | W)
          24 60018000 -> 00000000 (flags: R | W)
          25 60019000 -> 00000000 (flags: R | W)
          26 6001a000 -> 00000000 (flags: R | W)
          27 6001b000 -> 00000000 (flags: R | W)
          28 6001c000 -> 00000000 (flags: R | W)
          29 6001d000 -> 00000000 (flags: R | W)
          30 6001e000 -> 00000000 (flags: R | W)
          31 6001f000 -> 00000000 (flags: R | W)
          32 60020000 -> 40048000 (flags: VALID | R | W | USER | A | D)
          33 60021000 -> 40047000 (flags: VALID | R | W | USER | A | D)
          34 60022000 -> 40046000 (flags: VALID | R | W | USER | A | D)
          35 60023000 -> 4005a000 (flags: R | W | USER | A | D)
          36 60024000 -> 00000000 (flags: R | W)
     511 Superpage for 7fc00000 @ 40f08000 (flags: VALID)
        1019 7fffb000 -> 40f04000 (flags: VALID | R | W | USER | A | D)
        1020 7fffc000 -> 40f05000 (flags: VALID | R | W | USER | A | D)
        1021 7fffd000 -> 40f06000 (flags: VALID | R | W | USER | A | D)
        1022 7fffe000 -> 40f07000 (flags: VALID | R | W | USER | A | D)
        1023 7ffff000 -> 40f09000 (flags: VALID | R | W | USER | A | D)
     512 Superpage for 80000000 @ 40013000 (flags: VALID)
           0 80000000 -> 00000000 (flags: R | W)
           1 80001000 -> 00000000 (flags: R | W)
           2 80002000 -> 00000000 (flags: R | W)
           3 80003000 -> 00000000 (flags: R | W)
           4 80004000 -> 00000000 (flags: R | W)
           5 80005000 -> 00000000 (flags: R | W)
           6 80006000 -> 00000000 (flags: R | W)
           7 80007000 -> 00000000 (flags: R | W)
           8 80008000 -> 00000000 (flags: R | W)
           9 80009000 -> 00000000 (flags: R | W)
          10 8000a000 -> 00000000 (flags: R | W)
          11 8000b000 -> 00000000 (flags: R | W)
          12 8000c000 -> 00000000 (flags: R | W)
          13 8000d000 -> 00000000 (flags: R | W)
          14 8000e000 -> 00000000 (flags: R | W)
          15 8000f000 -> 00000000 (flags: R | W)
          16 80010000 -> 00000000 (flags: R | W)
          17 80011000 -> 00000000 (flags: R | W)
          18 80012000 -> 00000000 (flags: R | W)
          19 80013000 -> 00000000 (flags: R | W)
          20 80014000 -> 00000000 (flags: R | W)
          21 80015000 -> 00000000 (flags: R | W)
          22 80016000 -> 00000000 (flags: R | W)
          23 80017000 -> 00000000 (flags: R | W)
          24 80018000 -> 00000000 (flags: R | W)
          25 80019000 -> 00000000 (flags: R | W)
          26 8001a000 -> 00000000 (flags: R | W)
          27 8001b000 -> 00000000 (flags: R | W)
          28 8001c000 -> 00000000 (flags: R | W)
          29 8001d000 -> 40033000 (flags: VALID | R | W | USER | A | D)
          30 8001e000 -> 40032000 (flags: VALID | R | W | USER | A | D)
          31 8001f000 -> 40014000 (flags: VALID | R | W | USER | A | D)
    1021 Superpage for ff400000 @ 40f0b000 (flags: VALID)
           0 ff400000 -> 40f03000 (flags: VALID | R | W | A | D)
         256 ff500000 -> 40061000 (flags: VALID | R | W | A | D)
         384 ff580000 -> 40031000 (flags: VALID | R | W | A | D)
         511 ff5ff000 -> 40f08000 (flags: VALID | R | W | A | D)
         512 ff600000 -> 40013000 (flags: VALID | R | W | A | D)
        1021 ff7fd000 -> 40f0b000 (flags: VALID | R | W | A | D)
        1022 ff7fe000 -> 40f0c000 (flags: VALID | R | W | A | D)
    1022 Superpage for ff800000 @ 40f0c000 (flags: VALID)
           0 ff800000 -> 40f0d000 (flags: VALID | R | W | A | D)
           1 ff801000 -> 40f0a000 (flags: VALID | R | W | A | D)
    1023 Superpage for ffc00000 @ 40ee5000 (flags: VALID)
        <unavailable>
End of map

```
Note that I did have to make this change to get to this point:

https://github.com/betrusted-io/xous-core/blob/ea06df6879d740e33a1f30331c0c54aa25f41791/xous-ipc/src/buffer.rs#L136

Before the size was fixed to 4096 bytes. Uneven sizes are handled in the `new()` function by rounding the requested size up to the nearest whole page size; I did confirm this change worked for requests smaller than 4096 bytes.

You can reproduce the issue by running the `minimal` image. It will crash shortly after boot.

xobs, 2021-05-05:
This should be fixed in 23e759ffd8ff2ce26c9bacf5ba47a5cd979f3e4c

Please let me know if this solves the problem for you.

bunnie, 2021-05-05:
Looks like it fixes it, thanks!

closed 2021-05-05 by bunnie