betrusted-io/xous-core #76: x25519 vectors

bunnie, 2021-07-15:
Might be a good idea to port these into Xous and check the implementation against this set of vectors:

https://github.com/google/wycheproof/blob/master/testvectors/x25519_test.json

fweissberg, 2021-09-01:
I'd like to work on this issue and thought about implementing it this way:

* Add a new binary crate `wycheproof-import` with code to parse the test vectors and put them into a file `wycheproof_vectors.bin`.
    * Add the `x25519_test.json` to the crate for easy reproducibility and explicit versioning.
    * Potentially, only considering the `public`, `private` and `shared` fields of each test vector is sufficient since the result is always either `valid` or `acceptable`.
* Add an `wycheproof` subcommand to the `engine` `shellchat` command that runs the vectors using the methods from `x25519_dalek::{StaticSecret, PublicKey}` and reports the results as a summary.

Would that work for you? In particular, I was wondering if another `*.bin` file can be included without breaking something or using too much memory. If only the fields mentioned above were used, the file would contain 518 * 96 bytes.

bunnie, 2021-09-01:
Hello! Thank you for having a look at this. The good news is that @xobs has a port of engine25519 working in Renode, so you may even be able to emulate your solution to see if it works without the need for real hardware.

I think your approach works well. Overall, it makes sense to just include the vectors as a .bin file as long as there is a way to do that without having to invoke nightly Rust (iirc it's possible to do in stable, but I can't recall how it's done off the top of my head); alternatively, you could do something like an autogenerated Rust module that compiles into some static variables that can be accessed later on. This is the approach that was taken in absorbing the database of FPGA frames to patch for key injection into the bitstream: https://github.com/betrusted-io/xous-core/tree/main/services/root-keys/src/key2bits; there's a Python file that reads in the database and spits out an included .rs file. The main advantage of that approach is that the binary data gets type annotations around it without having to do some `unsafe` cast of a raw chunk of data to a slice or other structure. The main disadvantage is it takes forever for the linker to process big static structures like this.

From the memory usage standpoint, 50kiB of extra RAM dedicated to the test routine is not going to break the system. We're currently using about 6.7MiB out of 16MiB available RAM space (and all code is copied into RAM right now before being run). Eventually, we'll make deep diagnostic tests like this a "feature" option so we can save on code space, but for now we have enough memory space to carry all the test frameworks.

bunnie, 2021-09-01:
If you have any questions on how to get Redone running or how to navigate the code base, we welcome them. We're still figuring out what needs to be better documented to make the on-boarding process less painful.

fweissberg, 2021-09-01:
Thanks for the detailed feedback, it's much appreciated!

I think I found a way to use a .bin file without the need for nightly features by adapting what you were already doing with `engine25519_vectors.bin`. It will need some polishing but should be ready soon.

So far, I did not have any problems getting Xous running on Renode or navigating the code base. However, I was just gathering information for a bug report when @xobs  latest commits unbroke the system for me. Hosted mode worked fine, so I was able to play around a bit. Can't wait to see the hardware go into production. :)

closed 2021-09-06 by bunnie

bunnie, 2021-09-06:
@fweissberg has ported the vectors in, and the following failures are noted running on hardware:

```
INFO:engine_25519: montgomery job 0 (services\engine-25519\src\main.rs:823)
INFO:engine_25519: montgomery job 100 (services\engine-25519\src\main.rs:823)
INFO:engine_25519: montgomery job 200 (services\engine-25519\src\main.rs:823)
INFO:engine_25519: montgomery job 300 (services\engine-25519\src\main.rs:823)
INFO:engine_25519: montgomery job 400 (services\engine-25519\src\main.rs:823)
INFO:engine_25519: montgomery job 500 (services\engine-25519\src\main.rs:823)
ERR :shellchat::cmds::engine: wycheproof test #111 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: e5ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   f8ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #112 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: e3ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   f6ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #113 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: ddffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   f0ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #114 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: dbffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   eeffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff7f (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #131 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 46add6f48ffff461777d4f89b6fdf1155aa051a96387d45f3e5e371a236b6e52 (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   037fbaeb8f095241c1043eeb460946ec6fa0cf86121adb1db05bc9b35165c911 (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #132 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 1adbe32207e21f71e1af53884d2a2276481e298e557f4dacb3720f2458e3082d (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   83d1a2d110ced8f77e2db614eb8868ffe77e35e604191455b8e36bffc7876a78 (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #134 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 421bed1b26da1e9adbeada1f32b91a0fb4ced0f1110e0a4a88e735a19ee4571e (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   13a406639beee021ed01a21410106bdd9966841e1bc7bd74e7826b893eeae34b (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #135 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: d7b47463e2f4ca9a1a7deea098da8e74ac3b4a109083d997259b12992e7e7e06 (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   4f0fa6bb045d1d188ed74794a6564eba2734d25fb7fa8c753c557d95cf5c4618 (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #143 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 71133905b8a57ea8c38de0ecf213699a75b096c2df21f07f7e9eb03e9fa53f5c (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   5362024d256a73a8ebb04c3ea5ff68859d568a203a5e2e7d6d6657edd863f92c (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #144 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 3dc7b70e110766b2bf525252ebed98a100b2e532dc69544464da1bbab8625f6d (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   95c7df79b5d749df4005688f6d0ee3a6133bf32a664315269cc83cfc22053b31 (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #147 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: dfa988a477003be125b95ccbf2223d97729577d25e1d6e89e3da0afabdd0ae71 (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   18069892af1a48b2b94e6a83714425b978aea2c95772fe4c0d3d5f9ce391ea58 (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #161 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 4a7e2c5caf1d8180eb1c4f22692f29a14b4cdc9b193bd1d16e2f27438eef1448 (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   4e03046c63f1fe4f9f52d2532fd1ab757ea20ffd438884f1f75ae70810abb00a (services\shellchat\src\cmds\engine.rs:529)
ERR :shellchat::cmds::engine: wycheproof test #163 failed: (services\shellchat\src\cmds\engine.rs:527)
[... cut ...]
ERR :shellchat::cmds::engine: wycheproof test #507 failed: (services\shellchat\src\cmds\engine.rs:527)
ERR :shellchat::cmds::engine: expected: 0d8c21fa800ee63ce5e473d4c2975495062d8afa655091122cb41799d374594f (services\shellchat\src\cmds\engine.rs:528)
ERR :shellchat::cmds::engine: actual:   d86ec15f6ee8eaee3417ea7db0cd3aa3d7a2fa293be54320a053ab054564fc4d (services\shellchat\src\cmds\engine.rs:529)
```

The porting task is finished, but leaving this issue to track progress on fixing the failed tests.

bunnie, 2021-09-06:
The issue is fixed in this patch: https://github.com/betrusted-io/gateware/commit/817e284a3d92037b8cb0686735578d2bb60853e9

Specifically:
```patch
@@ -1413,7 +1413,7 @@ carries that have already been propagated. If we fail to do this, then we re-pro
                 self.dsp_match3 &
                 self.dsp_match2 &
                 self.dsp_match1 &
-                (self.dsp_p0 >= 0x1_ffed)
+                (self.dsp_p0[:17] >= 0x1_ffed)
             )
         ]
```
The normalization overflow computation was looking at the entire output of the DSP `p` register, which would cause a normalization to trigger when actually what we wanted was just a carry.

All vectors now pass with this patch.

Thanks @fweissberg for committing the test vectors that caught this very subtle bug!

PR #84 (2021-09-06): Port wycheproof vectors
Intended to close #76.

Please see wycheproof-import/README.md for a description.

Any feedback is most welcome.
commit 4bb454ef (2021-09-06): Import x25519 vectors from Project Wycheproof

Closes #76

The test vectors in wycheproof-import/x25519_test.json correspond to the upstream commit https://github.com/google/wycheproof/commit/4672ff74d68766e7785c2cac4c597effccef2c5c.

For details see wycheproof-import/README.md.