betrusted-io/xous-core #82: Panic and visual glitches when typing very long shellchat commands

samblenny, 2021-08-29:
Typing a very long shellchat command can trigger an array out of bounds panic at:

https://github.com/betrusted-io/blitstr/blob/cebcabb9a2b0686bb47f6e1508640941ae88469c/src/blit.rs#L46

To reproduce the panic, typing this string should be sufficient: "abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz012345".

The problem seems to happen when shellchat input wraps from 12 rows high to 13 rows high.

After panic, screen looks like this (note that suggestion line is blank at the bottom):

[image]

This PR to blitstr should resolve the panic: https://github.com/betrusted-io/blitstr/pull/2

But, when I run Xous with that patch, there are still some visual glitches with rendering very long shellchat commands. This is how it looked when I typed enough characters to wrap from 12 rows to 13 rows (with the blitstr patch):

[image]

This glitch did not cause a panic, and I was able to run the command (typing return echoed the long input string then triggered the help message).

bunnie, 2021-09-05:
The PR was absorbed into the system and it fixes the panic. I wasn't able to reproduce the visual glitch, however.

samblenny, 2021-09-06:
Sounds good enough to me. Should this be closed since the glitch is not reproducible?

bunnie, 2021-09-06:
I think i'd prefer it if you are agreeable to that. We can always open a new issue focusing on the glitch if we can reproduce it, but I suspect there may need to be a much larger refactor in order for long/large messages typed into the shellchat interface. I have noticed that the delay between keystroke acceptance grows as O(N) with message length and by the time you've filled the screen with text you're taking a very long time.

One thing I could really use some advice on is a way to extract the width of characters without having to make a full loop through the blitsr engine. Right now to compute the text wrapping, I do a "dry run" render of the characters, so everything short of blitting pixels. This is costly but fast enough for most purposes, but clearly, this is not going to cut it in the long term.

samblenny, 2021-09-06:
> One thing I could really use some advice on is a way to extract the width of characters without having to make a full loop through the blitsr engine. Right now to compute the text wrapping...

If you invoke blitstr with a ClipRect of the maximum width you want to use, it should handle the wrapping on its own. I'd recommend that you:
1. Allocate an offscreen framebuffer big enough to hold the largest text bubble you're willing to draw
2. Set a ClipRect within that offscreen buffer leaving enough padding on the top/left/right for drawing your left-aligned or right-aligned text bubble border
3. Invoke blitstr to render the string to your offscreen buffer
4. Check the cursor struct you passed as a `&mut` to blitstr to see how tall the text was (it will be wrapped to fit the ClipRect width, but height is variable based on line height and number of lines)
5. Use the dimensions of the text in the offscreen buffer to draw a text bubble border of the appropriate height in your main framebuffer
6. Blit the rendered text from the offscreen framebuffer into the text bubble
7. Keep the offscreen buffer around so you can use it if you need to redraw for a scroll (perhaps maintain a pool of ~5 such buffers that you can rotate through as needed to show the most recent commands and responses?)

closed 2021-09-06 by samblenny