betrusted-io/xous-core #91: Some `rkyv` suggestions / offer for help

djkoloski, 2021-11-23:
Hey all, sorry if an issue isn't the best way to get in contact.

I've been watching xous for a while and it's been super exciting how you're using rkyv. I saw https://github.com/betrusted-io/xous-core/commit/ce5120515518522cc3a3f22dafdc2afb0da728c6 recently and wanted to reach out to see if I could help alleviate some of the problems you're having with it. I'd like to make sure that you have the tools and support you need to make rkyv as effective as possible.

One specific recommendation I have would be to enable the `strict` feature if you want type layouts that are guaranteed not to change across minor versions (currently bugfix versions since major=0) and compiler upgrades. This effectively marks the generated types as `#[repr(C)]`, which I know you've been having some problems with but it may help.

In rkyv 0.6 serialization was also made fully deterministic by zeroing non-data bytes. We're all the way up to 0.7 now, and there are lots of new features available that might make working with the framework easier. Performance has also been improved considerably, so there might be something to be gained there as well. Upgrading across major version changes is obviously no small amount of work, so as always I'm here to help if I can.

If you're not already in it, we have [a discord channel](https://discord.gg/65F6MdnbQh) where you can swing by to get some more immediate help and provide feedback or ask questions. Good luck!

- David

bunnie, 2021-12-10:
Hi @djkoloski!

Sorry for the long lag on responding to this. Thank you very much for the comments. I didn't know about the `strict` feature -- that probably would have helped fix the issue I had encountered. We did end up using simple `#[repr(C)]` structures for the disk storage but this is less related to an rkyv issue and more to do with how the data layout interacts with the block encryption scheme applied at the layer underneath it -- all the stored keys have to be efficiently decomposed to fit into the encrypted blocks, and that decomposition process is outside the scope of something like `rkyv`, so we couldn't have effectively used it anyways.

That being said, it *is* on our list of things to do to upgrade our version of `rkyv`. I've toyed with it a couple of times and every time I've poked at it, I've retreated because it became too much work, and the system we have "works well enough" for now (to confess, I can't make heads or tails of the Serialization macros; so, I'll have to decompose Xous into a limited set of test cases to make sure I thoroughly understand the new API before doing the big upgrade).

So, I've been vaguely watching progress over in `rkyv` land and waiting for the velocity of major revs to slow down before doing the big upgrade. Eventually, I think we'd like to upgrade to a mature/stable version and then pin it there for the long-term. `xous` isn't meant to compete on providing a rich set of features: it's aiming for stability and auditability, so pinning a core library like `rkyv` and leaving it there, perhaps just applying patches only for security issues, is more in line with our long-term project's goals.

djkoloski, 2021-12-10:
Thanks for closing the loop, I appreciate it a lot. As long as you're having success with it, that's what matters. There are some cool new features that are on the way through the pipeline, and the next major revision should (hopefully) be the last major release before stabilizing for 1.0. Some of the final 1.0 goals for rkyv include a thorough security audit and stable format specification; I think those are probably the highest-value improvements for xous. Your goals make sense for your project, and I'm here to help if or when you decide to make the leap. 🙂 

Closing this for now!

closed 2021-12-10 by djkoloski

bunnie, 2021-12-10:
That's great to know! I think we're vaguely planning a "great refactoring" early next year, probably February-ish time frame. Not that I'm asking for any hard commitments, but just for our scheduling purposes do you think a 1.0 would have a snowball's chance in hell of being shipped by then? If it does, then that would be a really good candidate to take a good hard look at rkyv 1.0 and try to bring the codebase up to date for that.

djkoloski, 2021-12-10:
Haha, that would be insanely ambitious for rkyv so _probably_ not. There is a big holiday coming up though so I might find more time than expected to push on the milestone. That would also be for 0.8, the 0.8-1.0 gap is likely to be a year or more for stabilization.

bunnie, 2021-12-11:
glad I asked! that's very helpful to know for our long-term planning. I'll keep an eye out for the 0.8 tag, thanks for clarifying the timelines!