betrusted-io/xous-core #101: `ManagedMem` copies excess data from the framebuffer

xobs, 2022-02-14:
The `ManagedMem` primitive is used to save various regions when a suspend is performed.

The display is defined as:

```rust
pub struct XousDisplay {
    fb: MemoryRange,
    hwfb: MemoryRange,
    srfb: ManagedMem<{ utralib::generated::HW_MEMLCD_MEM_LEN }>,
    csr: utralib::CSR<u32>,
    susres: RegManager<{ utra::memlcd::MEMLCD_NUMREGS }>,
}
```

`HW_MEMLCD_MEM_LEN` is defined as `23584`. As a result, `srfb` gets assigned `4 * 23584` = 94336 bytes of data.

During a suspend, the manager loops through the target memory area which is rounded up to the nearest page:

```rust
    fn suspend(&mut self) {
        let src = self.mem.as_ptr() as *const u32;
        for words in 0..(self.mem.len() / core::mem::size_of::<u32>()) {
            self.backing[words] = unsafe{src.add(words).read_volatile()};
        }
    }
```

As a result, the program reads outside of the framebuffer when it stores data into the buffer.

In practice this isn't a problem since the backing store is 4x larger, and hardware ignores the reads outside of the buffer. However, it should probably read only enough data to fill the buffer.

xobs, 2022-02-14:
Can you see if this fixes both problems?

```patch
diff --git a/services/graphics-server/src/backend/betrusted.rs b/services/graphics-server/src/backend/betrusted.rs
index dc3aa294..7e5e877b 100644
--- a/services/graphics-server/src/backend/betrusted.rs
+++ b/services/graphics-server/src/backend/betrusted.rs
@@ -14,7 +14,7 @@ const CONFIG_CLOCK_FREQUENCY: u32 = 100_000_000;
 pub struct XousDisplay {
     fb: MemoryRange,
     hwfb: MemoryRange,
-    srfb: ManagedMem<{ utralib::generated::HW_MEMLCD_MEM_LEN }>,
+    srfb: ManagedMem<{ utralib::generated::HW_MEMLCD_MEM_LEN / 4 }>,
     csr: utralib::CSR<u32>,
     susres: RegManager<{ utra::memlcd::MEMLCD_NUMREGS }>,
 }
diff --git a/services/susres/src/api.rs b/services/susres/src/api.rs
index 44ad630d..8cdc7389 100644
--- a/services/susres/src/api.rs
+++ b/services/susres/src/api.rs
@@ -227,13 +227,13 @@ impl<const N: usize> ManagedMem<N> {
 impl<const N: usize> SuspendResume for ManagedMem<N> {
     fn suspend(&mut self) {
         let src = self.mem.as_ptr() as *const u32;
-        for words in 0..(self.mem.len() / core::mem::size_of::<u32>()) {
+        for words in 0..(N / core::mem::size_of::<u32>()) {
             self.backing[words] = unsafe{src.add(words).read_volatile()};
         }
     }
     fn resume(&mut self) {
         let dst = self.mem.as_ptr() as *mut u32;
-        for words in 0..(self.mem.len() / core::mem::size_of::<u32>()) {
+        for words in 0..(N / core::mem::size_of::<u32>()) {
             unsafe{dst.add(words).write_volatile(self.backing[words])};
         }
     }
```

bunnie, 2022-02-15:
Is this pushed to a branch so I can check it out and try it?

bunnie, 2022-02-15:
hmm, i guess I can just type in the changes.

bunnie, 2022-02-15:
looks good so far, I'm putting the device into a suspend/resume stress test loop and will close & merge if it passes

bunnie, 2022-02-15:
looks good, merged and pushed

closed 2022-02-15 by bunnie