betrusted-io/xous-core #104: Renode Errors during compilation or loading:

nworbnhoj, 2022-02-21:
I followed Wiki instructions to [Quickstart using an emulator](https://github.com/betrusted-io/xous-core#quickstart-using-an-emulator) and encountered "Errors during compilation or loading".

At the terminal:
`john@ubuntu:~/git/xous-core$ renode emulation/xous-release.resc
16:47:33.5133 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
16:47:40.1579 [INFO] Including script: /home/john/git/xous-core/emulation/xous-release.resc (3)
16:47:40.7578 [INFO] System bus created.
`

A Renode window opens with:
`Renode, version 1.12.0.37823 (44d6786a-202104022100)
(monitor) i $CWD/emulation/xous-release.resc
Current 'PATH' value is: /opt/renode;/home/john/git/xous-core;/home/john/git/xous-c
ore/emulation
Errors during compilation or loading:
Tried to load assembly that has been already loaded. Aborting operation.
Errors during compilation or loading:
There were compilation errors:
/home/john/git/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(12,39): error CS
0234: The type or namespace name `NORFlash' does not exist in the namespace `Antmic
ro.Renode.Peripherals.SPI'. Are you missing an assembly reference?
/home/john/git/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(144,63): error C
S0246: The type or namespace name `DecodedOperation' could not be found. Are you mi
ssing an assembly reference?
/home/john/git/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(596,17): error C
S0246: The type or namespace name `DecodedOperation' could not be found. Are you mi
ssing an assembly reference?
There was an error executing command 'machine LoadPlatformDescription @/home/john/g
it/xous-core/emulation/soc/betrusted-soc.repl'
Error E04: Could not resolve type: 'SPI.MXIC_MX66UM1G45G'.
At /home/john/git/xous-core/emulation/soc/betrusted-soc.repl:120:8:
flash: SPI.MXIC_MX66UM1G45G @ spinor
       ^^^^^^^^^^^^^^^^^^^^^
(SoC) 
`

xobs, 2022-02-21:
It looks like the `DecodedOperation` enum is not available in your version of Renode. Can you please try downloading the nightly version of Renode and seeing if that fixes things for you? https://dl.antmicro.com/projects/renode/builds/

nworbnhoj, 2022-02-21:
That works! Well done. Perhaps a note in the Wiki is in order?

bunnie, 2022-03-29:
@nworbnhoj can you suggest a page that would be the most logical place to look for such a note?

I'm not sure where people are going first to look for this information, and my interval view is quite different from the external view.

nworbnhoj, 2022-04-03:
I simply followed the "Download Renode" link under the heading [Quickstart using an emulator](https://github.com/betrusted-io/xous-core#quickstart-using-an-emulator) to download v1.12.0. So I guess you could redirect the link to the [nightly version](https://dl.antmicro.com/projects/renode/builds/) temporarily, until the `DecodedOperation` enum is included in the next tagged release.

bunnie, 2022-04-03:
I have updated the readme to have an additional link to the nightly version.

bunnie, 2022-05-23:
I think since the wiki etc. is updated, I'll close this issue -- if it needs further attention, feel free to re-open and specify what actions should be taken.

closed 2022-05-23 by bunnie