betrusted-io/xous-core #114: Infinite loop on "Can't boot: No valid keys!"

msuiche, 2022-03-18:
https://github.com/betrusted-io/xous-core/blob/5c155587b65c3e1a1b5b12c2dfe501e1f1836dd6/loader/src/secboot.rs#L351

When the above loop is hit, how does one restart the phone?

bunnie, 2022-03-18:
This is the state where you've provisioned the root keys, but it's stuck trying to find a signed image?

Can you try one thing -- use a paperclip to hit the button inside the hole on the lower right hand side of your device. 

It's possible that the FPGA didn't pick up the new root keys and so there's a mismatch between what's in the SoC and what's on the disk.

Failing that, then you'd need to do a factory reset to clear the SoC back to its default of looking for developer signed keys.

blakesmith, 2022-03-18:
I got stuck in this state as well. Steps that got me stuck here:

1. With a fully booted Precursor, I went to "Initialize Root Keys"
2. I accidentally pressed "enter" twice for each of the password prompts (I got confused by the 3-button dialogs that wanted me to pick my password hide mode), presumably setting 'blank' passwords for both keys.
3. Everything signed to completion.
4. Upon reboot, I got the "Can't boot: No valid keys!"

I tried to reproduce again, from a factory reset state, but was unsuccessful.

To get out of this state:

1. I used the "factory_reset.sh" script, and reflashed all the gateware / firmware (not sure if this helped anything). The screen stayed frozen with "Can't boot: No valid keys!" throughout the flash process.
2. Hit the paperclip reset button on the side, and Xous came back up with the initial developer keys again.

mguthaus, 2022-03-18:
Mine did this after I had initialized the root keys and did an update. It was fine after the paperclip.

bunnie, 2022-03-19:
Alright, it sounds like the gateware isn't automatically being reloaded properly. The device is supposed to reset itself, but it has to do a "cold" boot to pick up the latest gateware -- this could be hampered if the device is plugged into USB when that happens. If you're plugged into USB the best the device can do is a warm boot, unless you use the paperclip.

bunnie, 2022-03-30:
I think this issue is basically resolved with the paperclip, but leaving it open until the first tranche of users receive their device and get stuck on this.

uhtu, 2022-05-07:
just encountered this updating via update_ci.sh, happily this was the first hit for "precursor can't boot no valid keys"
and post-paperclip, the device boots