betrusted-io/xous-core #115: add default text API for `modals` Text entry

bunnie, 2022-03-18:
the `get_text()` API should get an `Option<String>` to have some default text populated in the dialog box:

```
    pub fn get_text(&self,
        prompt: &str,
        maybe_validator: Option<fn(TextEntryPayload, u32) -> Option<ValidatorErr>>,
        maybe_validator_op: Option<u32>,
    ) -> Result<TextEntryPayload, xous::Error> {
```

This will require some upstream changes to the actual TextEntry primitive itself to support this, so it's not a quick fix but it's a good idea.

bunnie, 2022-03-30:
Alright, this is slightly more complicated than I had thought because we'd need to add some state logic that deletes the default text if a backspace is hit or a character is input. I'm going to put this on the backburner for now as there are higher priority issues to chase after.

bunnie, 2022-03-30:
Dropping a help wanted label on this, in case anyone wants to try and tackle it.

gsora, 2022-04-05:
I have a somewhat working PoC [here](https://github.com/gsora/xous-core/tree/feature/multi-text-modals), along with a multiline-input enabled `get_text` method.

My Rust is *extremely* janky right now, reviews appreciated!

This is how it appears right now:

https://user-images.githubusercontent.com/145909/161804476-576a9a87-4daf-4380-b60b-58b59565f33e.mov

Labels are optional so in a multiline situation you are free to provide placeholders or not - in a single-line context the dot emoji is not shown.

The `modals` API has been changed slightly:

 - `get_text` accepts a `Option<String>` argument to account for the placeholder presence in the text view;
 - a new method `get_text_multi` has been added, which resembles `get_text` except it also accepts a `u32` (amount of fields requested in the modal) and a `Option<Vec<Option<String>>>` (placeholders for the text views);

`Option<Vec<Option<String>>>` is needed because a developer might want to:
 - not add placeholders;
 - only add a few placeholders;

bunnie, 2022-04-06:
this looks pretty cool! wow. I don't think I see anything I'd immediately tweak just browsing through the commit log.

bunnie, 2022-04-06:
as mentioned in chat, i'm about to do a refactor of `modals` for issue #111, so, apologize in advance for a potentially messy merge :(

bunnie, 2022-04-23:
This is fixed by @gsora in https://github.com/betrusted-io/xous-core/pull/140. closing.

closed 2022-04-23 by bunnie