betrusted-io/xous-core #118: Cannot connect to all 2.4Ghz wifi networks

chrisportela, 2022-03-18:
Sorry if this isn't the right place to be putting this, but I am guessing that it's a Xous problem and not a hardware problem (at least for now).

I'm using ec commit `8b5b8e50`, wf200 3.12.3, soc 0.9.5+0, dna `0x72452a193ea85c`. I haven't updated the device since receiving it yesterday.

I tried scanning for my wifi network which runs both 2.4 and 5Ghz networks using the same name but it never comes up. If I run my iPhone's hotspot with "maximum compatibility" (Unclear what this really means) then the precursor can connect.

Am I running into issues with the Wifi chip or is there work that needs to be done to the wifi stack? Are these known issues? Possibly already fixed in a newer version?

I tried looking for some docs in the wiki to see if there are some kind of logs I could see and try to understand what is happening with the wifi. I was able to do something yesterday that actually soft-bricked the device as it got stuck trying to do something with the wifi.

bunnie, 2022-03-18:
what do you get with 

`ver ec`
`ver xous`
`ver soc`

There were significant changes in the wifi stack. Soc 0.9.5 sounds like you might be running the stock firmware that came from the factory, so actually those version commands might not even exist :zap: 

If that's the case, please do try the update -- the wifi stack was in a terrible state at 0.9.5, it's not even worth trying to diagnose...

bunnie, 2022-03-18:
Yes, please give it a try. Make sure you're using a firmly seated USB cable -- during the factory reset procedure, if the cable falls loose, there is a chance you can't recover from that using USB, and you'd have to go back to failsafe. A couple users have already reported this but fortunately they have recovered using a Raspberry Pi.

chrisportela, 2022-03-19:
Updating using `factory-reset.ps1` on Win 11 worked. Took a few attempts to get it going after update (also, unclear, but I think I need to have root keys initialized for things to work?) but I managed to set ssid/pass and save it and then it joined properly.

I did notice a weird thing that saying `wlan join` after `wlan save` kept telling me to save, but then I checked `wlan status` and I was connected.

Also until I entered everything in I still wasn't seeing my network, but after entering it in + saving it, it showed up in the list of networks. That seems strange, but maybe my router is doing some thing weird.

bunnie, 2022-03-20:
Right - the `save` command is not automatic if you joined successfully. So you can see that you've got a good `wlan status` but the system doesn't assume you should save the credentials automatically. Maybe it should. But, unfortunately, the way it is written, you have to explicitly save them.

The rationale was that, sometimes it's hard to tell the difference between a joined-and-good network, and a joined-and-slightly misconfigured network. Maybe we could add a ping test or something and if it passes it saves the credentials. But right now, we leave that up to the user's judgement.

So it's not your router, this is...basically our UX fault. The `connection_manager` doesn't run and automatically keep you connected to your router until you've done a `wlan save`.

bunnie, 2022-03-20:
I _think_ this issue is resolved, so I'll close it - go ahead and re-open it if you need more help and feel free to add comments regardless.

closed 2022-03-20 by bunnie