betrusted-io/xous-core #125: cargo xtask run downloading non-existent file

philips, 2022-03-28:
I git cloned this repo and tried to run `cargo xtask run` but I got this error:

```

No apps specified, adding default apps...
Outdated toolchain installed.                                             
                                     
Error: Toolchain for riscv32imac-unknown-xous-elf was not found on this system!     
Would you like this program to attempt to download and install it?   [Y/n] y
Attempting to install toolchain for riscv32imac-unknown-xous-elf into /usr/local/Cellar/rust/1.58.1
Downloading from https://github.com/betrusted-io/rust/releases/latest/download/riscv32imac-unknown-xous_1.58.1.zip...
Download progress: 100.00%      
Download successful. Total data size is 9 bytes                                                                                                     
Extracting toolchain to /usr/local/Cellar/rust/1.58.1...                
Error: "unable to extract zip: invalid Zip archive"  
```

The problem is that https://github.com/betrusted-io/rust/releases/latest/download/riscv32imac-unknown-xous_1.58.1.zip is no longer a valid URL because https://github.com/betrusted-io/rust/releases/latest is now `1.59.0` not `1.58.1`. 

I don’t really understand all of the version parsing going on so here is a stupid patch that “fixes” it.

```
$ git diff 
diff --git a/xtask/src/main.rs b/xtask/src/main.rs
index 7db3f7bc..46514d6b 100644
--- a/xtask/src/main.rs
+++ b/xtask/src/main.rs
@@ -1167,7 +1167,7 @@ fn ensure_compiler(
 
     let toolchain_url = format!(
         "{}{}.{}.{}{}",
-        TOOLCHAIN_URL_PREFIX, ver.major, ver.minor, ver.patch, TOOLCHAIN_URL_SUFFIX
+        TOOLCHAIN_URL_PREFIX, 1, 59, 0, TOOLCHAIN_URL_SUFFIX
     );
 
     println!(
```

philips, 2022-03-29:
Updating the wiki page: https://github.com/betrusted-io/betrusted-wiki/wiki#hello-world

It makes sense to me that the build should just error out if rust is not at the exact right version instead of failing on fetching dependencies. Easier to debug and clearer for new folks.

bunnie, 2022-03-29:
@xobs What do you think? I'm ok with it but I'm not sure how to implement. My first thought would be to just hard-code a minimum supported rust version requirement in `xtask` as a `const` somewhere, but I'm not sure if there's a better way to do that.

xobs, 2022-03-29:
This was added in Rust 1.56.0, so we can specify a `rust-version` in the root `Cargo.toml`. Perhaps we can do something like this, which is completely untested:

```toml
[target.'cfg(target_os = "xous")'.package]
rust-version = "1.59"
```

If it does what I think it'd do, it would set the minimum supported rust version to 1.59 when targeting the "xous" OS.

xobs, 2022-03-29:
I've added this in the `xtask-ureq-assets` branch which adds MSRV, uses `ureq`, and uses the assets file instead of blindly fetching the URL:

 https://github.com/betrusted-io/xous-core/pull/new/xtask-ureq-assets

closed 2022-03-29 by bunnie

philips, 2022-03-29:
> that looks great. I just merged it and built it on my end to test it out, and merged it into main from the command line
>
> I'll go ahead and close this!

Thanks!

Sent from a pocket terminal