betrusted-io/xous-core #135: Wrong keyboard layout in hosted mode (debian 11)

bousares, 2022-04-07:
The keyboard works everywhere else but when I try to type commands in xous I mostly get symbols instead of letters. Many keys don't register anything. This is on a fresh installation of debian 11 with US keyboard with Rust version 1.59.0 (9d1b2106e 2022-02-23). Tried with gnome-terminal and Xterm.

Me pressing random letters on keboard:

```
INFO:keyboard: injecting key '-'(2d) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key '&'(26) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key '$'(24) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key ' '(20) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key '!'(21) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key '%'(25) (services/keyboard/src/main.rs:885)
INFO:keyboard: injecting key '#'(23) (services/keyboard/src/main.rs:885)
```

```
$ setxkbmap -print -verbose 10

Setting verbose level to 10
locale is C
Trying to load rules file ./rules/evdev...
Trying to load rules file /usr/share/X11/xkb/rules/evdev...
Success.
Applied rules from evdev:
rules:      evdev
model:      pc105
layout:     us
Trying to build keymap using the following components:
keycodes:   evdev+aliases(qwerty)
types:      complete
compat:     complete
symbols:    pc+us+inet(evdev)
geometry:   pc(pc105)
xkb_keymap {
	xkb_keycodes  { include "evdev+aliases(qwerty)"};
	xkb_types     { include "complete"	};
	xkb_compat    { include "complete"	};
	xkb_symbols   { include "pc+us+inet(evdev)"	};
	xkb_geometry  { include "pc(pc105)"	};
};
``
![Screenshot from 2022-04-07 04-02-24](https://user-images.githubusercontent.com/69337153/162133650-5ff468ae-9f86-4aa9-8a3c-5e397660877b.png)
`

bousares, 2022-04-07:
Tried the Renode emulator mode.

`$ cargo xtask renode-image`
...builds without error.

`$ renode emulation/xous-release.res`
...opens two windows. The window "Versatile:sysbus.uart0" is just black with a blinking cursor. The second window, "Renode", gives me the following error message:

```
Renode, version 1.12.0.37823 (44d6786a-202104022100)
(monitor) i $CWD/emulation/xous-release.resc
Current 'PATH' value is: /home/jo/bin;/home/jo/code/test/xous-core;/home/jo/code/test/xous-core/emulation
Errors during compilation or loading:
There were compilation errors:
/home/jo/code/test/xous-core/emulation/peripherals/ComEc.cs(105,35): error CS0246: The type or namespace name `IComPeripheral' c
ould not be found. Are you missing an assembly reference?
Errors during compilation or loading:
There were compilation errors:
/home/jo/code/test/xous-core/emulation/peripherals/ComSoC.cs(20,77): error CS0246: The type or namespace name `IComController' c
ould not be found. Are you missing an assembly reference?
/home/jo/code/test/xous-core/emulation/peripherals/ComSoC.cs(115,30): error CS0115: `Antmicro.Renode.Peripherals.SPI.BetrustedSo
cCom.Reset()' is marked as an override but no suitable method found to override
Errors during compilation or loading:
There were compilation errors:
/home/jo/code/test/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(12,39): error CS0234: The type or namespace name `NORFlas
h' does not exist in the namespace `Antmicro.Renode.Peripherals.SPI'. Are you missing an assembly reference?
/home/jo/code/test/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(144,63): error CS0246: The type or namespace name `Decode
dOperation' could not be found. Are you missing an assembly reference?
/home/jo/code/test/xous-core/emulation/peripherals/MXIC_MX66UM1G45G.cs(596,17): error CS0246: The type or namespace name `Decode
dOperation' could not be found. Are you missing an assembly reference?
There was an error executing command 'machine LoadPlatformDescription @/home/jo/code/test/xous-core/emulation/soc/betrusted-soc.
repl'
Error E04: Could not resolve type: 'SPI.BetrustedSocCom'.
At /home/jo/code/test/xous-core/emulation/soc/betrusted-soc.repl:35:6:
com: SPI.BetrustedSocCom @ sysbus 0xf000d000
     ^^^^^^^^^^^^^^^^^^^^
```
So there we have it. If anyone wants to spend time on this, please do so but you can also just ignore this because I'm thinking of giving up and test and see if I can get this thing to fly on Ubuntu instead.

xobs, 2022-04-08:
Regarding the Renode issue, you may not be using the Nightly builds, which are required due to some refactoring of various classes. Antmicro are working on releasing a new version, however their currently-released build is more than one year out of date.

Nightly binaries may be obtained from https://dl.antmicro.com/projects/renode/builds/

bunnie, 2022-04-08:
The code that handles character input from the `minifb` layer is very thin in hosted mode. Basically, we take the raw characters and pass them directly on without any lookup. There used to be a mapping step but that's largely been deprecated, it's only used now for swapping out delete/backspace and adding functionality to the home key.

https://github.com/betrusted-io/xous-core/blob/3ef2dae2f0a79b9b65afa716664f43063802774f/services/graphics-server/src/backend/minifb.rs#L262-L270

So, if there is a mapping issue, it's likely a problem in the `minifb` crate. I did a quick search there and didn't see any issues open that match yours, unfortunately...

bunnie, 2022-04-21:
I think https://github.com/betrusted-io/xous-core/pull/138 should resolve this. If it does, please close the issue.

vmedea, 2022-05-10:
This does sound like the same problem I had on Wayland. But this mentions X11 which makes me less certain.

bunnie, 2022-09-20:
No response and I think #138 resolved this. Closing the issue; if it's still a problem please re-open and provide more details.