betrusted-io/xous-core #141: missing `log-server` messages

bunnie, 2022-04-22:
When printing successive "large" messages to the `log-server`, one of them can go missing.

I have created a minimal test case that captures this condition. This is the core of the test case:

https://github.com/betrusted-io/xous-core/blob/75a07f8e1e995756f1e48ad58e1bb75a364052d8/services/test-stub3/src/main.rs#L62-L89

It prints a "large" `buf`, then lends the buffer; upon return, it prints the returned buffer (`buf_ret`). It then also prints an unrelated, second buffer `buf2. 

The "correct" case is committed here:
https://github.com/betrusted-io/xous-core/commit/75a07f8e1e995756f1e48ad58e1bb75a364052d8
I have created some `xtask msgtest` for your convenience, you should be able to just check out and run that to reproduce at that commit.

This case works because the length of the slice printed is truncated to 32 bytes. The output looks like this:

```
INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:98)
INFO:test_stub3: buf: [0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7] (services\test-stub3\src\main.rs:62)
INFO:test_stub3: small message 1 (services\test-stub3\src\main.rs:70)
INFO:test_stub3: buf_ret: [1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8] (services\test-stub3\src\main.rs:84)
INFO:test_stub3: buf2: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5] (services\test-stub3\src\main.rs:88)
INFO:test_stub3: small message 3 (services\test-stub3\src\main.rs:89)
INFO:usb_test: keepalive 1 (services\usb-test\src\main.rs:98)
INFO:usb_test: keepalive 2 (services\usb-test\src\main.rs:98)
INFO:test_stub3: buf: [0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7] (services\test-stub3\src\main.rs:62)
INFO:test_stub3: small message 1 (services\test-stub3\src\main.rs:70)
INFO:test_stub3: buf_ret: [1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8] (services\test-stub3\src\main.rs:84)
INFO:test_stub3: buf2: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5] (services\test-stub3\src\main.rs:88)
INFO:test_stub3: small message 3 (services\test-stub3\src\main.rs:89)
```

As you can see' there is a `buf_ret` printout in the log.

The "broken" case is committed here:
https://github.com/betrusted-io/xous-core/commit/bdae2218e6127a9ddec687d749a811601589ccf0
I have confirmed the error exists in both hosted mode and in renode. You can build a renode image using
`cargo xtask msgtestrenode`.

This is what the output looks like, slightly redacted for clarity:
```

INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:98)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:44)
INFO:test_stub3: buf: [0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2,
...
    , 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 0
     (services\test-stub3\src\main.rs:63)
     INFO:test_stub3: small message 1 (services\test-stub3\src\main.rs:71)
     INFO:test_stub3: buf2: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3
...
	 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5
	  (services\test-stub3\src\main.rs:89)
	  INFO:test_stub3: small message 3 (services\test-stub3\src\main.rs:90)
	  INFO:usb_test: keepalive 1 (services\usb-test\src\main.rs:98)
```

As you can see here, the `buf_ret` has gone missing.

I think the problem is related to, as @xobs mentioned, the fact that the `log-server` was implemented ages ago and has a potential buffer re-use issue. But, could you please take a look at this @xobs and confirm the situation?

Thanks to @nworbnhoj for originally reporting this bug.

xobs, 2022-04-23:
The good news is that this wasn't a kernel bug. It was a bug in the logging server implementation, and it was an off-by-one in the length calculation.

The way the log-server works is that it serializes the data into an `args` buffer that it then writes to the serial port.

There are checks to ensure that the various fields don't exceed their array sizes. These used `>=` instead of `>`, meaning that maximum-sized arrays were getting dropped:

https://github.com/betrusted-io/xous-core/blob/main/services/log-server/src/main.rs#L328

What appears to be happening is that the log server rejects messages that are exactly the full length of the buffer. You can "fix" the test program by adding one character to the message which causes the `BufferWrapper` to not pad it to a full size of the buffer:

```
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: small message 1 (services\test-stub3\src\main.rs:73)
INFO:test_stub3: buf_ret: [1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10
, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4
, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9
, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 
3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 
8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11,
 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5,
 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10
, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4
, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9
, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 
3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 
8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11,
 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5,
 7, 9, 11, 2, 4, 6, 8, 10, 1, 3, 5, 7, 9, 11, 2, 4, 6, 8 (services\test-stub3\src\main.rs:87)
INFO:test_stub3: buf22: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2,
 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5,
[... cut ...]
 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5, 2, 6, 3, 0, 4, 1, 5 (services\test-stub3\src\main.rs:91)
INFO:test_stub3: small message 3 (services\test-stub3\src\main.rs:92)
```

The patch for this was added in 234c169809bb3e0ce649406ad0a17ed8e0d346d3.

Regardless, I've gone ahead and refactored the log server. It now uses the same idiom of stack-allocating messages. I've also increased the log message size so that it's now 3820 bytes instead of 3000 bytes. The new refactor is present in f0f1b72e0c8dba542cb0b9aaafe6294a17c75535.

Please let me know if these changes solve the issue you're seeing.

bunnie, 2022-04-23:
yes, this fixes my test case. thank you! I'll go ahead and close this and add this to the release notes.

closed 2022-04-23 by bunnie