betrusted-io/xous-core #142: Different errors with different renode nightly builds

degregat, 2022-04-22:
I've tried three different of the most recent nightly builds of renode and they give me different errors when trying to start the renode-image. Can someone post an exact version or even better, a link to a known good build?

If it is of interest I can reproduce and paste the error messages, or even automate testing the different nightly versions to find which ones work or not, but it seems to be different quirks of the different builds and not something systemic related to xous.

Thank you very much :)

xobs, 2022-04-23:
Hi @degregat , can you please post the error messages you're getting?

degregat, 2022-04-23:
It was my fault sorry. I did not run `renode` from the decompressed directory, so it couldn't find `libllvm-disas.so`.

For the record, this was the error message the latest nightly threw:

`
Fatal error:
Could not load type of field 'Antmicro.Renode.Disassembler.LLVM.LLVMDisassembler:<Disassemble>k__BackingField' (0) due to: Could not resolve type with token 01000010 from typeref (expected class 'Antmicro.Renode.Peripherals.CPU.Disassembler.DisassemblyProvider' in assembly 'Renode-peripherals, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null') assembly:Renode-peripherals, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null type:Antmicro.Renode.Peripherals.CPU.Disassembler.DisassemblyProvider member:(null)
`

closed 2022-04-23 by degregat