betrusted-io/xous-core #144: "regifting" problem

bunnie, 2022-04-23:
As I had mentioned previously, I had some troubles with "regifting" a memory buffer. This is the basic idiom:

https://github.com/betrusted-io/xous-core/blob/b084cb675c5fe5c31e749b6f5f71c13556992ade/services/usb-test/src/main.rs#L165-L180

The idea is that you have a memory message coming in, and you want to forward it on to a process-local thread that you want to keep private because you don't want to expose all its APIs as some are sensitive.

In this working example, we get this as the output:

```
INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:140)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
INFO:usb_test: first level rx (services\usb-test\src\main.rs:166)
INFO:usb_test: regift! (services\usb-test\src\main.rs:172)
INFO:usb_test: incoming second hand message (services\usb-test\src\main.rs:89)
INFO:usb_test: got [42, 0, 42, 0], 100 (services\usb-test\src\main.rs:94)
INFO:usb_test: keepalive 1 (services\usb-test\src\main.rs:140)
INFO:usb_test: keepalive 2 (services\usb-test\src\main.rs:140)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
```

It works as expected. Here is the version that doesn't work:
https://github.com/betrusted-io/xous-core/blob/6185e5e3dfea3eb816b0afa3bc0d26b2a0b419fa/services/usb-test/src/main.rs#L165-L180

Here, I'm taking the `buffer` that is derived from the `MemoryMessage` and directly doing a `send()` on it again, without first copying it onto the stack. It might be that this is simply the wrong way to do regifting, or it could be a bug. I suspect I'm just doing it wrong, so, I've marked this as a `question` and not `bug` for now.

This is what the output looks like in Renode for the "broken" run:

```
INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:140)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
INFO:usb_test: first level rx (services\usb-test\src\main.rs:166)
INFO:usb_test: regift! (services\usb-test\src\main.rs:172)
INFO:usb_test: incoming second hand message (services\usb-test\src\main.rs:89)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
```

And in hosted mode it's a bit more dramatic:
```
INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:140)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
INFO:usb_test: first level rx (services\usb-test\src\main.rs:166)
INFO:usb_test: regift! (services\usb-test\src\main.rs:172)
KERNEL(7): client disconnected: An existing connection was forcibly closed by the remote host. (os error 10054) -- shutting down virtual process
KERNEL(7): Finished the thread so sending TerminateProcess
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
KERNEL(8): client disconnected: An existing connection was forcibly closed by the remote host. (os error 10054) -- shutting down virtual process
KERNEL(8): Finished the thread so sending TerminateProcess
Panic Details: PanicInfo { payload: Any { .. }, message: Some(PID 8 does not exist), location: Location { file: "kernel\\src\\arch\\hosted\\process.rs", line: 85, col: 17 }, can_unwind: true }
```

Something gets a bit upset with the regifting process and simply quits, resulting eventually in a panic because a PID ceases to exist. This is the PID table in hosted mode, fwiw:

```
  PID  |  Command
-------+------------------
   2   |  ..\target\release\ticktimer-server
   3   |  ..\target\release\log-server
   4   |  ..\target\release\susres
   5   |  ..\target\release\xous-names
   6   |  ..\target\release\trng
   7   |  ..\target\release\usb-test
   8   |  ..\target\release\test-stub3
   ```

This is not a show-stopper because the workaround is to simply copy the data before passing it on. But, since @xobs mentioned this should be within the semantics of a `MemoryMessage`, I figured it was worth creating a test case at least to be looked at.

xobs, 2022-04-23:
Does it work if you do the following after you do `buffer.send()`?:

```rust
core::mem::forget(buffer)
```

bunnie, 2022-04-23:
`buffer.send()` moves `buffer` out of scope so on the sender if you try to do that, you get the following compilation error:

```
error[E0382]: use of moved value: `buffer`
   --> services\usb-test\src\main.rs:180:35
    |
167 |                 let buffer =
    |                     ------ move occurs because `buffer` has type `Buffer<'_>`, which does not implement the `Copy` trait
...
173 |                 buffer.send(  // <-- note buf -> buffer change
    |                 ------ value moved here
...
180 |                 core::mem::forget(buffer);
    |    
    ```

xobs, 2022-04-23:
What about:

 ```rust
core::mem::forget(msg);
```

bunnie, 2022-04-23:
compiles, runs, but still fails. hosted mode example:
```
INFO:usb_test: keepalive 0 (services\usb-test\src\main.rs:140)
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
INFO:usb_test: first level rx (services\usb-test\src\main.rs:166)
INFO:usb_test: regift! (services\usb-test\src\main.rs:172)
KERNEL(7): client disconnected: An existing connection was forcibly closed by the remote host. (os error 10054) -- shutting down virtual process
KERNEL(7): Finished the thread so sending TerminateProcess
INFO:test_stub3: test-stub3 pinger (services\test-stub3\src\main.rs:46)
INFO:test_stub3: initial send (services\test-stub3\src\main.rs:60)
KERNEL(8): client disconnected: An existing connection was forcibly closed by the remote host. (os error 10054) -- shutting down virtual process
KERNEL(8): Finished the thread so sending TerminateProcess
Panic Details: PanicInfo { payload: Any { .. }, message: Some(PID 8 does not exist), location: Location { file: "kernel\\src\\arch\\hosted\\process.rs", line: 85, col: 17 }, can_unwind: true }
```

xobs, 2022-04-24:
Is there a nice, reproducible test case I could see so that I can work on this? Hosted mode is fine here.

bunnie, 2022-04-24:
did you run it as `cargo xtask msgtest`? it's the same tooling and repos as the log server issue

xobs, 2022-04-24:
I did, but I'm not seeing the crash. I'm trying `cargo xtask msgtest` on revision bdae2218e6127a9ddec687d749a811601589ccf0 with this patch applied:

```patch
diff --git a/services/net/src/main.rs b/services/net/src/main.rs
index be3e6080..ef98f8ef 100644
--- a/services/net/src/main.rs
+++ b/services/net/src/main.rs
@@ -1642,9 +1642,9 @@ fn xmain() -> ! {
                 let buffer =
                     unsafe { Buffer::from_memory_message(msg.body.memory_message().unwrap()) };
                 // have to transform it through the local memory space because you can't re-lend pages
-                let sub = buffer.to_original::<WifiStateSubscription, _>().unwrap();
-                let buf = Buffer::into_buf(sub).expect("couldn't convert to memory message");
-                buf.send(
+                // let sub = buffer.to_original::<WifiStateSubscription, _>().unwrap();
+                // let buf = Buffer::into_buf(sub).expect("couldn't convert to memory message");
+                buffer.send(
                     cm_cid,
                     connection_manager::ConnectionManagerOpcode::SubscribeWifiStats
                         .to_u32()
```

bunnie, 2022-04-24:
There's a version committed that should require no edits to get the failure: 6185e5e3dfea3eb816b0afa3bc0d26b2a0b419fa

bunnie, 2022-04-25:
Fixed with PR #151 thanks @xobs !

closed 2022-04-25 by bunnie