betrusted-io/xous-core #145: Picking the currently running app in the app switching menu causes the current app to stop rendering.

andrewtron3000, 2022-04-23:
When I run xous in hosted mode on my computer with the ball demo:

`cargo xtask run ball`

I can switch to the ball app and see the ball bouncing around. However, when I use the menu to "Switch to App" and switch to the same Ball app in the menu, the ball app returns to the screen but the ball seems to be frozen and not be updating.

There is discussion of this issue in the betrustred matrix channel chat:

[image]

https://matrix.to/#/!NsnVDrDBdLREhyjNGd:matrix.org/$v4eHNMOPnqqiUX9GH5i3rtbN2j3sgEWfv-VMkGgIIRw?via=matrix.org&via=tchncs.de&via=aria-net.org

bunnie, 2022-04-23:
Thanks for opening this. I took a look at it and my gut was wrong, here's the explanation that I inserted in the code patch:

```
                        // the `activate()` method on `context_mgr` does not handle app-to-app switches when you're the same app.
                        // this is because there is a rule which says if you're going to put an app on top of an app, take the old
                        // app and hide it, while also taking the new app and showing it. This is a race condition, and it results
                        // in the "same-app" switch ending up in a hidden state, because within the routine the entering state is
                        // set before the leaving state due to interior-mutability issues (see the block under the debug statement
                        // titled "resolving visibility rules"). The fix is to simply not issue a context switch if it's the same app.
                        // It saves CPU thrashing and also works around this problem. (see issue #145)
```

If this fixes it for you, please close the issue.

andrewtron3000, 2022-04-24:
Looks like this patch fixes the issue.  I can now switch to the same app and the app continues.

closed 2022-04-24 by andrewtron3000