betrusted-io/xous-core #159: Usage of crossbeam channels leads to `ticktime-server` panic or watchdog reboot

gsora, 2022-05-20:
I'm working on a utility/server which wakes up backlight for 10 seconds, and keeps it alive until the user stops pressing hardware buttons (or UART key input) --- code is [here](https://github.com/gsora/xous-core/tree/gsora/backlight-on-keypress).

To do this, I used crossbeam channels.

If I put Precursor to sleep and then wake it up there's a chance of getting a panic from `ticktime-server`, or (as @bunnie suggested [here](https://matrix.to/#/!NsnVDrDBdLREhyjNGd:matrix.org/$bX_QPHT9etzMOB9nIkwhtiH-XhYEiicG7qZ8Ra0kry8?via=matrix.org&via=aria-net.org&via=beeper.com)) the server locking up but not hitting any panic instruction leads to a watchdog-induced reboot.

UART logs show the following:

```
WARNING: Tried to wait on a message that didn't exist
PANIC in PID 10: panicked at 'index out of bounds: the len is 128 but the index is 4294967295', kernel/src/services.rs:1814:50
PROGRAM HALT: CPU Exception on PID 10: Instruction page fault of 0xffd0a4f8 at 0xffd0a4fa
```

Attaching kernel debug with `console kernel` show the following:

```
WARNING: Tried to wait on a message that didn't exist
PROGRAM HALT: CPU Exception on PID 8: Store page fault of 0x7ffdeff8 at 0x0002d652
Current thread 0:
Thread 0:
PC:0002d652   SP:7ffdefc0   RA:0002d512
GP:00000000   TP:00000000
T0:7ffdf1b8   T1:0000002a   T2:00000000
T3:00000000   T4:00000000   T5:00000000   T6:00000000
S0:00000003   S1:0000013f   S2:7ffdf108   S3:00000001
S4:00014714   S5:00000000   S6:00014b68   S7:00000000
S8:00000000   S9:00000000  S10:00000000  S11:00000000
A0:00000000   A1:fffffc03   A2:7ffdf1d8   A3:00014714
A4:00000001   A5:00035828   A6:00000001   A7:00000000
Memory Maps for PID 8:
       0 Superpage for 00000000 @ 40978000 (flags: VALID)
          16 00010000 -> 40ded000 (flags: VALID | R | X | USER | A | D)
          17 00011000 -> 40dec000 (flags: VALID | R | X | USER | A | D)
          18 00012000 -> 40deb000 (flags: VALID | R | X | USER | A | D)
          19 00013000 -> 40dea000 (flags: VALID | R | X | USER | A | D)
          20 00014000 -> 40de9000 (flags: VALID | R | X | USER | A | D)
          21 00015000 -> 40de8000 (flags: VALID | R | X | USER | A | D)
          22 00016000 -> 40de7000 (flags: VALID | R | X | USER | A | D)
          23 00017000 -> 40de6000 (flags: VALID | R | X | USER | A | D)
          24 00018000 -> 40de5000 (flags: VALID | R | X | USER | A | D)
          25 00019000 -> 40de4000 (flags: VALID | R | X | USER | A | D)
          26 0001a000 -> 40de3000 (flags: VALID | R | X | USER | A | D)
          27 0001b000 -> 40de2000 (flags: VALID | R | X | USER | A | D)
          28 0001c000 -> 40de1000 (flags: VALID | R | X | USER | A | D)
          29 0001d000 -> 40de0000 (flags: VALID | R | X | USER | A | D)
          30 0001e000 -> 40ddf000 (flags: VALID | R | X | USER | A | D)
          31 0001f000 -> 40dde000 (flags: VALID | R | X | USER | A | D)
          32 00020000 -> 40ddd000 (flags: VALID | R | X | USER | A | D)
          33 00021000 -> 40ddc000 (flags: VALID | R | X | USER | A | D)
          34 00022000 -> 40ddb000 (flags: VALID | R | X | USER | A | D)
          35 00023000 -> 40dda000 (flags: VALID | R | X | USER | A | D)
          36 00024000 -> 40dd9000 (flags: VALID | R | X | USER | A | D)
          37 00025000 -> 40dd8000 (flags: VALID | R | X | USER | A | D)
          38 00026000 -> 40dd7000 (flags: VALID | R | X | USER | A | D)
          39 00027000 -> 40dd6000 (flags: VALID | R | X | USER | A | D)
          40 00028000 -> 40dd5000 (flags: VALID | R | X | USER | A | D)
          41 00029000 -> 40dd4000 (flags: VALID | R | X | USER | A | D)
          42 0002a000 -> 40dd3000 (flags: VALID | R | X | USER | A | D)
          43 0002b000 -> 40dd2000 (flags: VALID | R | X | USER | A | D)
          44 0002c000 -> 40dd1000 (flags: VALID | R | X | USER | A | D)
          45 0002d000 -> 40dd0000 (flags: VALID | R | X | USER | A | D)
          46 0002e000 -> 40dcf000 (flags: VALID | R | X | USER | A | D)
          47 0002f000 -> 40dce000 (flags: VALID | R | X | USER | A | D)
          48 00030000 -> 40dcd000 (flags: VALID | R | X | USER | A | D)
          49 00031000 -> 40dcc000 (flags: VALID | R | X | USER | A | D)
          50 00032000 -> 40dcb000 (flags: VALID | R | X | USER | A | D)
          51 00033000 -> 40dca000 (flags: VALID | R | X | USER | A | D)
          52 00034000 -> 40dc9000 (flags: VALID | R | X | USER | A | D)
          53 00035000 -> 40dc8000 (flags: VALID | R | W | X | USER | A | D)
[... 85 lines cut ...]
        1020 7fffc000 -> 4031d000 (flags: VALID | R | W | USER | A | D)
        1021 7fffd000 -> 400ed000 (flags: VALID | R | W | USER | A | D)
        1022 7fffe000 -> 400ee000 (flags: VALID | R | W | USER | A | D)
        1023 7ffff000 -> 4097a000 (flags: VALID | R | W | USER | A | D)
    1021 Superpage for ff400000 @ 4097c000 (flags: VALID)
           0 ff400000 -> 40978000 (flags: VALID | R | W | A | D)
         128 ff480000 -> 40012000 (flags: VALID | R | W | A | D)
         384 ff580000 -> 40057000 (flags: VALID | R | W | A | D)
         511 ff5ff000 -> 40979000 (flags: VALID | R | W | A | D)
        1021 ff7fd000 -> 4097c000 (flags: VALID | R | W | A | D)
        1022 ff7fe000 -> 4097d000 (flags: VALID | R | W | A | D)
    1022 Superpage for ff800000 @ 4097d000 (flags: VALID)
           0 ff800000 -> 4097e000 (flags: VALID | R | W | A | D)
           1 ff801000 -> 4097b000 (flags: VALID | R | W | A | D)
    1023 Superpage for ffc00000 @ 408f2000 (flags: VALID)
        <unavailable>
End of map
Program suspended. You may inspect it using gdb.
```

PID 8 in this instance is `ticktime-server`.

@bunnie then suggests that there might be weirdness involved in calculations done with stored server indices [here](https://matrix.to/#/!NsnVDrDBdLREhyjNGd:matrix.org/$QVEuIRtLySuntR1YtHWrTMiFAJrlEYpOchpZJnXuDJw?via=matrix.org&via=aria-net.org&via=beeper.com).

There's not a 100%-hit reproducer for this issue, but usually it shows itself up after a couple cycles of sleep -> start.

bunnie, 2022-05-20:
Thanks for filing this @gsora , thank you very much for the detailed report!

bunnie, 2022-05-20:
How long after the resume does it take for the reboot or panic to happen, and/or does it seem that every time after S/R you get the problem, or is it like...1 out of 4 times?

gsora, 2022-05-20:
It usually happens as soon as the backlight goes on after pressing a key. Sadly it doesn’t *always* happen, but it eventually does.

bunnie, 2022-05-20:
But based on my reading of the bug, it will never happen if you don't put it through a suspend/resume cycle, correct? Which means there is also an interaction with the suspend/resume process happening here.

gsora, 2022-05-20:
Correct, my previous comment was implying at least a sleep/resume cycle.

bunnie, 2022-05-20:
Can you try to cherry-pick ed16fc1b2f4388b2797e196d9d92d7a00c3322f6 and see if it fixes your problem? After consulting the guru (@xobs) his meditation responds as follows:

> My theory is that it's queueing a connection to respond to when the timer expires, but if the timer expires twice then it tries to respond to a null connection id.
> And if it goes to suspend in the same millisecond as it services the condvar, then it'll reissue the interrupt when it comes back from suspend.

Funny enough, there was a line of code reserved for just this very edge case...

bunnie, 2022-05-20:
In light testing on my side, I have not been able to trigger the issue after 4 or 5 s/r cycles, but before the patch, it was reasonably easy to aggravate.

gsora, 2022-05-21:
Sadly the issue is still there 😞 I was able to reproduce it with a couple suspend/resume cycles. Serial logs look the same as before.

bunnie, 2022-05-21:
hmmm strange. I just ran a 100-cycle s/r stress test and saw no signs of issues after 100 cycles.

But, I am using the latest version of the SoC and I was doing this with a firmware rev that was based off of taking my `main` and applying patches to add the `keyboard-backlight` server. 

I guess the next step is for me to try to reproduce this on a unit running specifically your version of everything, including your SoC (part of the issue is my developer unit SoC is now the version that has USB device support, and it's not code-compatible with the older versions of Xous because some SoC registers were added to support USB device, shifting around some hardware registers).

I'll try that tomorrow!

bunnie, 2022-05-23:
I think this bug contains two bugs. The first one:
```
WARNING: Tried to wait on a message that didn't exist
PANIC in PID 10: panicked at 'index out of bounds: the len is 128 but the index is 4294967295', kernel/src/services.rs:1814:50
PROGRAM HALT: CPU Exception on PID 10: Instruction page fault of 0xffd0a4f8 at 0xffd0a4fa
```
I think is solved by the +1 to the ticktimer on suspend, to avoid re-triggering the interrupt on resume.

The panic log you found:
```
WARNING: Tried to wait on a message that didn't exist
PROGRAM HALT: CPU Exception on PID 8: Store page fault of 0x7ffdeff8 at 0x0002d652
Current thread 0:
```
Can be triggered without a suspend/resume cycle, and seems linked to condvar implementation itself, so I have opened a new bug #162.