betrusted-io/xous-core #162: Crossbeam channels leads to system lockup

bunnie, 2022-05-23:
This was discovered while fixing #159. I think #159 is actually fixed now with the +1 hack to the timer restore to prevent the double-interrupt on resume. Now, with more testing, we're finding a new corner case so this is opened in a new issue.

I think the core of the code at question looks like this:
https://github.com/betrusted-io/xous-core/blob/53ad7644324d0d0d5626638404aad7a05880b3b9/services/keyboard-backlight/src/main.rs#L104-L116

Basically, there is a channel that receives an enum. One of two things happen: (1) the timeout is set to (in this implementation) 5 seconds from now; (2) a timeout is set for that time in the future, and if that timeout happens, a message gets fired off.

The first `send` into the channel results in a condvar somewhere being set with a timeout trigger about 5 seconds in the future, as can be seen in the logs:

`INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200001ec with a timeout of 4998 add to table of 1 entries (services\ticktimer-server\src\main.rs:730)`

the second time you tap the keyboard before the timeout happens, the condvar is updated:
`INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200001ec with a timeout of 4999 add to table of 2 entries (services\ticktimer-server\src\main.rs:730)`

eventually, if you stop hitting the keyboard, the condvar retires:
```
INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200001ec with a timeout of 4994 add to table of 2 entries (services\ticktimer-server\src\main.rs:730)
TRCE:keyboard_backlight: timeout finished, total re-waited 228, returning! (services\keyboard-backlight\src\main.rs:111)
TRCE:keyboard_backlight: turning lights off (services\keyboard-backlight\src\main.rs:81)
```

then on the next key hit, it seems the table gets another entry:
`INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200002d4 with a timeout of 4998 add to table of 3 entries (services\ticktimer-server\src\main.rs:730)`

If you keep mashing on the keyboard, eventually you get something like this:
```
INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200002d4 with a timeout of 4988 add to table of 3 entries (services\ticktimer-server\src\main.rs:730)
PANIC in PID 12:
thread 'main' panicked at 'assertion failed: `(left == right)`
                                                                left: `1`,
                                                                           right: `0`', /home/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/hashbrown-0.12.1/src/raw/mod.rs:2144:13
```
or in the kernel log, maybe I'll get a "WARNING: Tried to wait on a message that didn't exist", followed by a reboot, or a panic like this:
```
WARNING: Tried to wait on a message that didn't exist
PROGRAM HALT: CPU Exception on PID 8: Store page fault of 0x7ffdefac at 0x0002d538
Current thread 0:
Thread 0:
PC:0002d538   SP:7ffdefa0   RA:0002dbe0
GP:00000000   TP:00000000
T0:7ffdf0f8   T1:0000002a   T2:00000000
T3:00000000   T4:00000000   T5:00000000   T6:00000000
S0:7ffdf118   S1:0000013f   S2:7ffdf048   S3:00000001
S4:0001474c   S5:00000000   S6:00014ba0   S7:00000000
S8:00000000   S9:00000000  S10:00000000  S11:00000000
A0:7ffdf048   A1:00014ba0   A2:7ffdf118   A3:0001474c
A4:00000001   A5:00035870   A6:00000001   A7:00000000
Memory Maps for PID 8:
       0 Superpage for 00000000 @ 40946000 (flags: VALID)
          16 00010000 -> 40dea000 (flags: VALID | R | X | USER | A | D)
          17 00011000 -> 40de9000 (flags: VALID | R | X | USER | A | D)
          18 00012000 -> 40de8000 (flags: VALID | R | X | USER | A | D)
          19 00013000 -> 40de7000 (flags: VALID | R | X | USER | A | D)
          20 00014000 -> 40de6000 (flags: VALID | R | X | USER | A | D)
          21 00015000 -> 40de5000 (flags: VALID | R | X | USER | A | D)
          22 00016000 -> 40de4000 (flags: VALID | R | X | USER | A | D)
          23 00017000 -> 40de3000 (flags: VALID | R | X | USER | A | D)
          24 00018000 -> 40de2000 (flags: VALID | R | X | USER | A | D)
          25 00019000 -> 40de1000 (flags: VALID | R | X | USER | A | D)
...
   511 Superpage for 7fc00000 @ 40947000 (flags: VALID)
         991 7ffdf000 -> 405a4000 (flags: VALID | R | W | USER | A | D)
         992 7ffe0000 -> 405a3000 (flags: VALID | R | W | USER | A | D)
         993 7ffe1000 -> 405a2000 (flags: VALID | R | W | USER | A | D)
         994 7ffe2000 -> 405a1000 (flags: VALID | R | W | USER | A | D)
         995 7ffe3000 -> 405a0000 (flags: VALID | R | W | USER | A | D)
         996 7ffe4000 -> 4059f000 (flags: VALID | R | W | USER | A | D)
         997 7ffe5000 -> 4059e000 (flags: VALID | R | W | USER | A | D)
         998 7ffe6000 -> 4059d000 (flags: VALID | R | W | USER | A | D)
         999 7ffe7000 -> 4059c000 (flags: VALID | R | W | USER | A | D)
        1000 7ffe8000 -> 4059b000 (flags: VALID | R | W | USER | A | D)
        1001 7ffe9000 -> 4059a000 (flags: VALID | R | W | USER | A | D)
        1002 7ffea000 -> 40599000 (flags: VALID | R | W | USER | A | D)
        1003 7ffeb000 -> 40598000 (flags: VALID | R | W | USER | A | D)
        1004 7ffec000 -> 40597000 (flags: VALID | R | W | USER | A | D)
        1005 7ffed000 -> 40596000 (flags: VALID | R | W | USER | A | D)
        1006 7ffee000 -> 40595000 (flags: VALID | R | W | USER | A | D)
        1007 7ffef000 -> 40594000 (flags: VALID | R | W | USER | A | D)
        1008 7fff0000 -> 40593000 (flags: VALID | R | W | USER | A | D)
        1009 7fff1000 -> 40592000 (flags: VALID | R | W | USER | A | D)
        1010 7fff2000 -> 40591000 (flags: VALID | R | W | USER | A | D)
        1011 7fff3000 -> 40590000 (flags: VALID | R | W | USER | A | D)
        1012 7fff4000 -> 4058f000 (flags: VALID | R | W | USER | A | D)
        1013 7fff5000 -> 4058e000 (flags: VALID | R | W | USER | A | D)
        1014 7fff6000 -> 4058d000 (flags: VALID | R | W | USER | A | D)
        1015 7fff7000 -> 4058c000 (flags: VALID | R | W | USER | A | D)
        1016 7fff8000 -> 4058b000 (flags: VALID | R | W | USER | A | D)
        1017 7fff9000 -> 4058a000 (flags: VALID | R | W | USER | A | D)
        1018 7fffa000 -> 40589000 (flags: VALID | R | W | USER | A | D)
        1019 7fffb000 -> 400f2000 (flags: VALID | R | W | USER | A | D)
        1020 7fffc000 -> 40588000 (flags: VALID | R | W | USER | A | D)
        1021 7fffd000 -> 400f0000 (flags: VALID | R | W | USER | A | D)
        1022 7fffe000 -> 400f1000 (flags: VALID | R | W | USER | A | D)
        1023 7ffff000 -> 40948000 (flags: VALID | R | W | USER | A | D)
```

This is the process table for reference:
```
        PID 1: kernel
        PID 2: gam
        PID 3: status
        PID 4: shellchat
        PID 5: ime-frontend
        PID 6: ime-plugin-shell
        PID 7: graphics-server
        PID 8: ticktimer-server
        PID 9: log-server
        PID 10: com
        PID 11: xous-names
        PID 12: keyboard
        PID 13: keyboard-backlight
        PID 14: trng
        PID 15: llio
        PID 16: susres
        PID 17: codec
        PID 18: sha2
        PID 19: engine-25519
        PID 20: spinor
        PID 21: root-keys
        PID 22: jtag
        PID 23: net
        PID 24: dns
        PID 25: pddb
        PID 26: modals
        PID 27: usb-device-xous
```

bunnie, 2022-05-23:
After instrumenting the condvar routines, it seems that we have a problem with entries in the waiting vector not being removed:
```
INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200002d4 with a timeout of 4996 add to table of 3 entries (services\ticktimer-server\src\main.rs:730)
INFO:ticktimer_server: New waiting senders: Some([Sender { data: 218759221 }]) (services\ticktimer-server\src\main.rs:775)
TRCE:keyboard_backlight: timeout finished, total re-waited 11, returning! (services\keyboard-backlight\src\main.rs:112)
TRCE:keyboard_backlight: turning lights off (services\keyboard-backlight\src\main.rs:82)

INFO:keyboard_backlight: SPAWN NEW THREAD (services\keyboard-backlight\src\main.rs:70)
INFO:ticktimer_server: sender in pid Some(13) is waiting on a condition 200002d4 with a timeout of 4998 add to table of 3 entries (services\ticktimer-server\src\main.rs:730)
INFO:ticktimer_server: New waiting senders: Some([Sender { data: 218759221 }, Sender { data: 218759230 }]) (services\ticktimer-server\src\main.rs:775)
```

So here, we are printing the sender ID of the waiting senders. The order of events are:

1. Thread starts, and sends out a condvar
2. Condvar retires the thread, and the thread quits
3. New thread is spawned, with the same thread ID, and incidentally, the same condvar address -- but with a different message sender ID
4. Condvar pushes this new message ID, but somehow, the old message ID is still there.
5. Things crash, because a message is returned to the old message ID.

I think the problem can be traced to an implementation issue in `RecalculateSleep`. This commit I think fixes it: 8221f92d954a57931023556582ecc8e76f408db2

Here is the original RecalculateSleep:

https://github.com/betrusted-io/xous-core/blob/3bfdc238a060f18410ac1c31d478d6dd5afd828a/services/ticktimer-server/src/main.rs#L628-L651

And here is the revised version:

https://github.com/betrusted-io/xous-core/blob/8221f92d954a57931023556582ecc8e76f408db2/services/ticktimer-server/src/main.rs#L625-L649

The key difference is that in the original routine, the `sender_pid` is being calculated from the sender of the actual `RecalculateSleep` message -- which is always, by definition, the `ticktimer_server`. Thus the code later on never gets around to removing the entry inside the `notify_hash` structure, so it persists even after the notification is sent.

The proposed patch extracts the `sender_pid` from the `sender` argument, and with this, the correct entry seems to be removed.

However, I really want @xobs to have a look at this before merging the change, because there is a very explicit check here:

https://github.com/betrusted-io/xous-core/blob/3bfdc238a060f18410ac1c31d478d6dd5afd828a/services/ticktimer-server/src/main.rs#L632

That seems to very much want to make sure that the `sender_pid` is actually equal to that of the ticktimer_server (xous::process::id() will never return anything other than that, anyways) -- so I think it's more likely that I'm not understanding how this routine works, since what I think is a "bug" is actually being checked effectively with an assert later on.

I'm guessing there is a subtlety I'm missing here...

bunnie, 2022-05-23:
On the topic of "how did this every work before" if this is actually the bug, I think the reason why things didn't go south before is that in none of the code paths typically used before was a condvar re-used for a thread that was destroyed and subsequently re-created. So the fact that the entry persisted between calls was harmless, because the sender ID for a given PID always remained the same. That would by my theory, at least, for why this even worked in the first place and passed tests?

Essentially the edge case is destorying a thread and creating a new one that contains a condvar that happens to have the same address, but in all previous tests, they were done either in a brand new thread, or the re-allocated thread had a different address for the condvar so it wouldn't re-use the vector containing its message sender IDs.

bunnie, 2022-05-23:
After 596 suspend/resume stress test cycles, I finally did see one error. I don't have much to go on other than this:

```
INFO:shellchat::cmds::sleep: fuzz: 332/927 (services\shellchat\src\cmds\sleep.rs:89)
'(d) (services\keyboard\src\main.rs:904)
TRCE:keyboard_backlight: renewing backlight timer (services\keyboard-backlight\src\main.rs:62)
PANIC in PID 8:
thread 'main' panicked at 'couldn't send response: ProcessNotFound', services\ticktimer-server\src\main.rs:818:55
```
It corresponds to this code (offset by one line because it's a different branch, but the actual code is the same):

https://github.com/betrusted-io/xous-core/blob/3bfdc238a060f18410ac1c31d478d6dd5afd828a/services/ticktimer-server/src/main.rs#L807-L818

it's the return on line 817 here that's faliing.

This is a different path than the bug that was fixed in the PoC above -- @xobs could you please just eyeball that code and let me know if you can think if any problems with it? I feel like what could have happened?

It shouldn't be possible that `RecalculateSleep` happens in the middle of the access to `notify_hash` and the stop sleep call, because this is within the same thread. But...should every entry have a `return_scalar()` call on it, or should the entry only be returned if `v.sender == entry`?

xobs, 2022-05-23:
The code is checking that the sender is the ticktimer server only as a security mechanism -- that way only the ticktimer server is allowed to request a requeue. This also allows us to keep the ownership of the ticktimer array within the main function rather than having it global and wrapped in a mutex.

Also, it's probably not an issue if the address is reused. Because they're wrapped in an `Arc` the actual value is meant to be shared across threads.

However, there may be an issue where the `Arc` count goes to zero while the timeout hasn't expired.

One thing we can try doing is creating an index that's an `AtomicUsize`. Add an additional value to https://github.com/betrusted-io/rust/blob/1.61.0-xous/library/std/src/sys/xous/locks/condvar.rs#L10 that's monotonically increasing. Every time `new()` is called at https://github.com/betrusted-io/rust/blob/1.61.0-xous/library/std/src/sys/xous/locks/condvar.rs#L20 you increment this value.

Then, pass that index as the second argument in all functions instead of `self.counter.as_mut_ptr() as usize`.

That way if a condvar gets destroyed while a timeout is pending, then recreated at the same address, there won't be a race condition with a reused index.

xobs, 2022-05-23:
I've attempted an experiment in bda895dd9880ba4aab20bed7f73f1dc3bbeb639f where we use a monotonically increasing ID for the index instead of the raw pointer address. The ID is some arbitrary value, so anything is fine.

Previously we were using the address in order to ensure it could be used in a `const` function, however it seems as though the `.init()` call allows us to use an initializer, which grants us the ability to have monotonic identifiers.

xobs, 2022-05-24:
Correct, the core fix is:

https://github.com/betrusted-io/xous-core/blob/3bfdc238a060f18410ac1c31d478d6dd5afd828a/services/ticktimer-server/src/main.rs#L636-L637

This should read:

https://github.com/betrusted-io/xous-core/blob/8221f92d954a57931023556582ecc8e76f408db2/services/ticktimer-server/src/main.rs#L635-L636

We were taking the wrong PID, which ended up being fine most of the time since we never actually timed out. There were a few cases where this happened in testing, but I believe the numbers were so small that we ended up using default values that worked anyway.

We did merge a change in `libstd` that serialized condvars, causing them to be monotonically increasing rather than based on pointers. However, that is a separate enhancement.

bunnie, 2022-05-24:
Restored the security check for the origin of RecalculateSleep with 5815eae023147e29a75cd61fe336c1b8d98ce4dd

Patches have been pushed to main and tested -- closing issue. Thanks @xobs for the review and fixes!

closed 2022-05-24 by bunnie