betrusted-io/xous-core #163: Missing math functions when compiling

gsora, 2022-05-23:
While testing @nworbnhoj xous-core patch to enable image display (`dither` branch), I encountered an issue: truncf, floor and ceil functions are missing.

Code snippet:

```rust
            // 5. test image
            log::info!("testing image");
            let input = include_bytes!("tests/bunny.png");
            let image = match image::load_from_memory(input).and_then(|img| Ok(img.to_rgb8())) {
                Err(err) => {
                    log::error!("cannot load image, {}", err);
                    return;
                },
                Ok(img) => Img::new(
                    img.pixels().map(|p| RGB::from(p.0)),
                    img.width(),
                ),
            }.unwrap();

            modals.show_image(image).expect("image failed");
            log::info!("image test done");
```

Error:

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o" "--as-needed" "-L" "/Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps" "-L" "/Users/gsora/Documents/Code/nworbnhoj-xous-core/target/release/deps" "-L" "/Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/build/xous-bd123beb9668a9b1/out" "-L" "/Users/gsora/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/riscv32imac-unknown-xous-elf/lib" "-Bstatic" "/var/folders/f4/8plkl27x0d58bk_x7_zvqly40000gn/T/rustcTEuWXi/libxous-5d84267e68e07581.rlib" "--start-group" "/var/folders/f4/8plkl27x0d58bk_x7_zvqly40000gn/T/rustcTEuWXi/libxous-e71a2877b6337e51.rlib" "--end-group" "/Users/gsora/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/riscv32imac-unknown-xous-elf/lib/libcompiler_builtins-b47508d752ba9bd4.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "/Users/gsora/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/riscv32imac-unknown-xous-elf/lib" "-o" "/Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5" "--gc-sections" "-O1"
  = note: rust-lld: error: undefined symbol: floor
          >>> referenced by buffer.rs:0 (xous-ipc/src/buffer.rs:0)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(core::ops::function::FnOnce::call_once$u7b$$u7b$vtable.shim$u7d$$u7d$::h1360978f3480dd8d)
          
          rust-lld: error: undefined symbol: ceil
          >>> referenced by buffer.rs:0 (xous-ipc/src/buffer.rs:0)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(core::ops::function::FnOnce::call_once$u7b$$u7b$vtable.shim$u7d$$u7d$::h1360978f3480dd8d)
          >>> referenced by utils.rs:0 (/Users/gsora/.cargo/registry/src/github.com-1ecc6299db9ec823/png-0.16.8/src/utils.rs:0)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(png::utils::Adam7Iterator::init_pass::h800f4d57c73c2f04)
          >>> referenced by utils.rs:131 (/Users/gsora/.cargo/registry/src/github.com-1ecc6299db9ec823/png-0.16.8/src/utils.rs:131)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(png::utils::Adam7Iterator::init_pass::h800f4d57c73c2f04)
          
          rust-lld: error: undefined symbol: truncf
          >>> referenced by upsampler.rs:180 (/Users/gsora/.cargo/registry/src/github.com-1ecc6299db9ec823/jpeg-decoder-0.1.22/src/upsampler.rs:180)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(_$LT$jpeg_decoder..upsampler..UpsamplerH1V2$u20$as$u20$jpeg_decoder..upsampler..Upsample$GT$::upsample_row::h7f8fab5c85e25faa)
          >>> referenced by upsampler.rs:203 (/Users/gsora/.cargo/registry/src/github.com-1ecc6299db9ec823/jpeg-decoder-0.1.22/src/upsampler.rs:203)
          >>>               /Users/gsora/Documents/Code/nworbnhoj-xous-core/target/riscv32imac-unknown-xous-elf/release/deps/modals-112700aadd8508b5.log_server-f1b68f37d6a5fd75.12szj93pctdve1cu.rcgu.o.rcgu.o:(_$LT$jpeg_decoder..upsampler..UpsamplerH2V2$u20$as$u20$jpeg_decoder..upsampler..Upsample$GT$::upsample_row::h52374ef0c4cbf6f5)
          

error: could not compile `modals` due to previous error
```

The `image` crate must be added as a dependency of `modals` for this snippet to work:

```toml
image = { version = "0.23", default-features = false, features = ["png"] }
```

bunnie, 2022-05-23:
@xobs I think all we need to do is update the signatures here:

https://github.com/betrusted-io/rust/blob/1.61.0-xous/library/std/src/sys/xous/cmath.rs

as the ceil function is in libm already:

https://github.com/rust-lang/libm/blob/1f7b8eb61cab5f62ec93d2343432bebd1ada30f2/src/math/ceil.rs

but weirdly enough `compiler_builtins` seems to only have signatures for SGX?

https://github.com/rust-lang/compiler-builtins/blob/4117da3b2a97061730732041ba2ef0c1edbe0e53/src/math.rs#L96-L104

I am probably not understanding how this works tho -- I think you'd be able to weigh in super fast on how to do this. 

If it's just a matter of updating the signatures, maybe we should take all the latest ones and incorporate them? It looks like they have more intrtinsics available now than before. 

If it's not just a matter of adding the signatures, maybe we just copy/paste the stuff out of `libm` and stick it in Xous standard library?

bunnie, 2022-05-23:
I did a quick smoke test and `ceil()` did link for me in a test routine. @gsora if it works for you please go ahead and close the issue!

gsora, 2022-05-24:
Compiling works now, I'm closing this issue.

Thanks!

closed 2022-05-24 by gsora