betrusted-io/xous-core #165: Panic on "failed to forward Opcode"

nworbnhoj, 2022-05-26:
The [ws_issue_A](https://github.com/nworbnhoj/xous-core/tree/ws_issue_A) branch will Panic with shellchat `test ws`

```
PANIC!
Details: PanicInfo { payload: Any { .. }, message: Some(failed to forward Opcode::Open: (Envelope { sender: Sender { data: 68812800 }, body: MutableBorrow(MemoryMessage { id: 0, buf: MemoryRange { addr: 139804406714368, size: 4096 }, offset: Some(4), valid: Some(4096) }) }, MemoryInUse)), location: Location { file: "services/websocket/src/main.rs", line: 41, col: 22 } }
```

The panic can be averted by removing the following code from `manager.rs`

```
    let _ = match log_server::init_wait() {
        Ok(_) => {
            println!("log_server OK ");
            ()
        }
        Err(e) => println!("log_server Error {:#?}", e),
    };
```

nworbnhoj, 2022-05-26:
Furthermore - a panic can also be triggered in `manager.rs` with `log_server::init_wait().unwrap();`

```
PANIC!
Details: PanicInfo { payload: Any { .. }, message: Some(called `Result::unwrap()` on an `Err` value: ()), location: Location { file: "services/websocket/src/manager.rs", line: 14, col: 29 } }

```

bunnie, 2022-05-27:
are these panics on hardware or on hosted mode? I just tried building it and I got an error about rand_core compatibility for a hardware target, so I'm thinking maybe you're running this as hosted mode?

bunnie, 2022-05-27:
ah yes this is testing with the websockets stuff -- so it's hosted mode.

bunnie, 2022-05-27:
> Furthermore - a panic can also be triggered in `manager.rs` with `log_server::init_wait().unwrap();`
>
> ```
> PANIC!
> Details: PanicInfo { payload: Any { .. }, message: Some(called `Result::unwrap()` on [...]

For this one, the call to `log_server::init_wait()` -- it can only be done once per server (not thread). It's already done in your main function, so the second call does not work.

bunnie, 2022-05-27:
@xobs it looks like the `.forward()` method on `MessageEnvelope` is missing a case for handling the correct return type of a mutable lend.

I inserted a print here:

https://github.com/betrusted-io/xous-core/blob/51444925a930d7806045ec165e3238f465a96dfb/xous-rs/src/definitions/messages/envelope.rs#L124-L137

And the return type is

`result: Ok(MemoryReturned(Some(4), Some(4096)))`

This results in the function terminating in a `return Err((new_self, crate::Error::MemoryInUse));` statement.

I'm guessing maybe what was meant to have happened in that case was for the `MemoryReturned` result to just be bumped back to the caller, but I'm not confident to make the change because there are some fancy things going on with lifetimes here.

Mind having a look at this when you're feeling better? I suspect it's an easy one or two liner for you to fix. thanks!

xobs, 2022-06-09:
Can you try cherry-picking f4f54181901c38d4eebbcfd2beabb4f99fbda63e and see if that helps you?

bunnie, 2022-06-09:
fwiw i ran the test and it no longer panics

commit f4f54181 (2022-06-09): xous: envelope: handle MemoryReturned in forward()

Previously, we only handled messages that were forwarded. With this
patch, we also handle memory messages that were returned.

This fixes #165