betrusted-io/xous-core #167: `services/benchmark` and `apps/hello` fail to compile

xobs, 2022-05-28:
When building all packages in the repository, these two fail to compile. Removing them from the list of workspace members fixes the problem. Should these services get removed?

```
[6:11:12 am] E:/Code/Xous/Core> cargo build --target riscv32imac-unknown-xous-elf
   Compiling benchmark v0.1.0 (E:\Code\Xous\Core\services\benchmark)
   Compiling hello v0.1.0 (E:\Code\Xous\Core\apps\hello)
error[E0432]: unresolved import `blitstr`
 --> services\benchmark\src\main.rs:8:5
  |
8 | use blitstr::GlyphStyle;
  |     ^^^^^^^ use of undeclared crate or module `blitstr`

error[E0432]: unresolved import `blitstr_ref`
 --> services\benchmark\src\main.rs:9:5
  |
9 | use blitstr_ref as blitstr;
  |     ^^^^^^^^^^^^^^^^^^^^^^ no external crate `blitstr_ref`

error[E0425]: cannot find value `APP_NAME_HELLO` in crate `gam`
  --> apps\hello\src\main.rs:41:66
   |
41 |                 app_name: xous_ipc::String::<128>::from_str(gam::APP_NAME_HELLO),
   |                                                                  ^^^^^^^^^^^^^^ help: a constant with a similar name exists: `APP_NAME_BALL`
   |
  ::: E:\Code\Xous\Core\services\gam\src\apps.rs:2:1
   |
2  | pub const APP_NAME_BALL: &'static str = "ball demo app";
   | -------------------------------------------------------- similarly named constant `APP_NAME_BALL` defined here

For more information about this error, try `rustc --explain E0432`.
error: could not compile `benchmark` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
For more information about this error, try `rustc --explain E0425`.
error: could not compile `hello` due to previous error
[6:12:16 am] E:/Code/Xous/Core>
```

bunnie, 2022-05-29:
Benchmark has bit rotted, yah. I'll remove it from the workspace, but would like to keep the code around a reference because I have some numbers on the speed of Xous scalar and memory messages, and if we had to reproduce those benchmarks I'd start by unrotting that service.

Hello "should" work, not sure what happened. Will look into it.

bunnie, 2022-05-29:
Ah i see you're just using a generic 'build' command from the top without xtask

The apps won't build without xtask -- it dynamically generates the headers and structures for the apps based on which ones are included. So the missing `APP_NAME_HELLO` is generated by the xtask app-image script, if you select "hello" as one of the apps to build, by parsing the `manifest.json` file.

I think this is...not a bug?

bunnie, 2022-05-29:
Ah i think maybe the problem is that they are part of the default-members. I just removed the apps from the default-member set.

bunnie, 2022-05-29:
yah that seems to fix it. pushed.

Also, could you take a pass at the "stale" branches on `xous-core` that are owned by you and trim any that you think aren't needed anymore? I don't want to delete any of your branches because I don't know if you're using them or not, but I think it would be good to do a little housecleaning there.

xobs, 2022-05-29:
I've trimmed those branches now.

bunnie, 2022-05-29:
awesome, stick a fork in it i think this is done.

closed 2022-05-29 by bunnie