betrusted-io/xous-core #174: Dependency Policy possibly not being enforced?

Hoverbear, 2022-06-07:
I was reading https://github.com/betrusted-io/xous-core/commit/dd943c54163c36f98e715e3065585a549d6e7bea and wanted to make sure that the policy you desired was being enforced as you wanted.

Observing the [Book](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-cratesio ) (Emph added)

> Cargo is configured to look for dependencies on [crates.io](https://crates.io/) by default. Only the name and a version string are required in this case. In [the cargo guide](https://doc.rust-lang.org/cargo/guide/index.html), we specified a dependency on the time crate:
>
> ```toml
> [dependencies]
> time = "0.1.12"
> ```
>
> The string "0.1.12" is a version requirement. **Although it looks like a specific version of the time crate, it actually specifies a range of versions and allows [SemVer](https://semver.org/) compatible updates.** An update is allowed if the new version number does not modify the left-most non-zero digit in the major, minor, patch grouping. In this case, if we ran cargo update -p time, cargo should update us to version 0.1.13 if it is the latest 0.1.z release, but would not update us to 0.2.0. If instead we had specified the version string as 1.0, cargo should update to 1.1 if it is the latest 1.y release, but not 2.0. The version 0.0.x is not considered compatible with any other version.

[Later in the book](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#comparison-requirements) we can observe:

> Comparison requirements allow manually specifying a version range or an **exact version to depend on.**
>
> Here are some examples of comparison requirements:
>
> ```
> >= 1.2.0
> > 1
> < 2
> = 1.2.3
> ```

From these pieces of documentation one can conclude from the [commit comment](https://github.com/betrusted-io/xous-core/commit/dd943c54163c36f98e715e3065585a549d6e7bea):

> I'd like this process to be very deliberate and not automatic for Xous.

Perhaps the desire would be for dependencies to be `=1.0.0` instead of `1.0.0`?

bunnie, 2022-06-07:
The idea is that if someone were to pull the Xous repository and just run the build command, they should be able to reproduce our build as we have it on our machines. The packages used should match the versions we snapshotted using `crate-scraper` at the time a release was tagged, even if they were to try and go back to a repo state a year ago (let's say hypothetically -- for now this wouldn't work because rustc has changed versions, but let's imagine a world where we stopped chasing the latest rustc). 

My understanding is the "0.1.12" specifier would mean that someone just doing a generic "run" command would grab "0.1.12", and not a later version like "0.1.14". However, if someone ran "cargo update", then, yes, they would grab the "0.1.14" version and then their .lock file would differ from ours.

I guess maybe the question is -- would the upgrade to "0.1.14" only happen if `cargo update` is run? If it is the case that the upgrade can only happen if and only if someone deliberately ran `cargo upgrade` then I think I'm OK with that -- anyone who worried about reproducibility of a build would not issue that command, as they would know it should break things.

On the other hand, if `cargo update` implicit in any brand-new fetch/build of a fresh Xous repository, then we should be using the "=0.1.12" style of pinning down dependencies, because it would in fact be impossible for someone to reproduce our build anytime a minor version of an update is pushed. 

The main concern here is that anyone who wants to "build Xous themselves" because they don't trust us should be able to. We aren't there yet because OS path names are built into the binaries, but I think that is currently the only blocker. But if cargo just automatically grabs the latest version on a clean build then someone could push a supply chain attack in between one of our releases and the time a user tries to build Xous themselves, in which case they could not rely upon the results of an audit...that's the high level context, at least, for the requirement.

Do you happen to know which way cargo would behave given the above scenarios outlined? thanks!

Hoverbear, 2022-06-07:
> The idea is that if someone were to pull the Xous repository and just run the build command, they should be able to reproduce our build as we have it on our machines. 

Barring a [trusting trust](https://manishearth.github.io/blog/2016/12/02/reflections-on-rusting-trust/) attack or something similar, the existing `Cargo.lock` should be sufficient for that without requiring you to specify the minors (or an `=`) at all.

However if you seek to protect against a blind `cargo update` doing this `=1.0.0` can lock you to specific versions, forcing something like a `cargo outdated` to find updates and set them manually.

Since this repo ships a `Cargo.lock`, if you add some dep `boop = "1.0.0"` it may install "1.0.1" instead, but once that is committed to the lock file, that's what your other users will get (provided they don't use things as a library, or use them off crates.io, or wipe the lock).

If you want to avoid some `0.1.12` upgrading to a `0.1.14` on a `cargo update`, you'd want to do `=0.1.12`.

Typically end users would not do `cargo update in binary repositories like this. It's typically safe and ok for binaries to do things like `=0.1.12` as their dependencies, however in libraries you typically want to only depend on the lowest version (eg `"0.1.12"`) style unless you have special requirements, as this will result in the consuming binary being able to create a more minimal dependency set.

As for a supply chain attack, you should always commit and ship a `Cargo.lock`, this is sufficient to build in things like `nix` which are entirely reproducible.

If you're looking through dependencies I recommend `cargo tree` (notably the `cargo tree -i rand:0.7.0` style) and `cargo depgraph` (to make graphviz stuff)

Hoverbear, 2022-06-09:
I was thinking a bit more today, one thing you might consider is using `cargo vendor` on releases and providing those as artifacts. This would allow users to download that vendor'd dep package and use it in their own build, without depending on `crates.io`. This is useful for if you want to build offline, or if you are doing work in a remote area. It's also used in some Nix builders for Rust.

bunnie, 2022-06-10:
ok changed wording in e32ea4d3f5b17269853ade10abf512334bec26e3, closing. thanks!

commit e32ea4d3 (2022-06-10): fix #174