betrusted-io/xous-core #188: Add Syscall to increase heap on a per-process basis

bunnie, 2022-06-16:
It'd be nice to have a syscall to increase the heap limit of a given process. Right now the heap limit across all processes is 512kiB set by https://github.com/betrusted-io/xous-kernel/commit/11963414703ce4e6a22bcd7ea7f31d6d28635ca8.

In particular, it would be nice to increase just the PDDB's heap size because it will heap-allocate file cache structures. Ideally all the caching is done in the PDDB and not in user processes, so we just want to increase the PDDB's limit, and not everyone's across the board.

xobs, 2022-06-23:
Can you see if https://github.com/betrusted-io/xous-core/tree/adjust-process-limit does what you want?

In order to adjust a limit, you must first present the current limit. This prevents two threads from doing something unexpected. An example of how this might work:

```rust
fn main() {
    println!("Increasing our heap");
    println!("Reading current heap size...");
    let new_limit = 2048 * 1024;
    let result = xous::rsyscall(xous::SysCall::AdjustProcessLimit(
        xous::Limits::HeapMaximum as usize,
        0,
        new_limit,
    ));

    println!("Result of call: {:?}", result);
    if let Ok(xous::Result::Scalar2(1, current_limit)) = result {
        println!("Our previous heap was: {}", current_limit);
        xous::rsyscall(xous::SysCall::AdjustProcessLimit(
            xous::Limits::HeapMaximum as usize,
            current_limit,
            new_limit,
        ))
        .unwrap();
        println!("Our new heap is: {}", new_limit);
    } else {
        panic!("Unsupported syscall!");
    }
}
```

This produces the following output:

```
Increasing our heap
INFO:ticktimer_server: my PID is 2 (services\ticktimer-server\src\main.rs:522)
INFO:xous_names: started (services\xous-names\src\main.rs:386)
Reading current heap size...
Result of call: Ok(Scalar2(1, 524288))
INFO:ticktimer_server: Welcome to Xous v0.9.8-629-g08958745 (services\ticktimer-server\src\main.rs:525)
Our previous heap was: 524288
Our new heap is: 2097152
```

So far I've implemented limit calls for `HeapMaximum` and `HeapSize`. Note that `HeapSize` can be used instead of `IncreaseHeap` and `DecreaseHeap`, which makes those calls somewhat redundant.

If this addresses the issue, please merge it into main.

bunnie, 2022-06-26:
Well, it didn't work, and it worked. 

How it didn't work: I tried putting the call right at the top of `shellchat` initialization. When I do this, some *other* servers fail to start. I'm not sure why, but it looked like at least the root keys server was no longer able to connect. 

How it did work: i put the call just on top of the part where an image was downloaded in shellchat. For the first time ever, I now see this:

[image]

(^^ @nworbnhoj  ^^)

Looks like i needed a megabyte of heap to decode it:
```
INFO:shellchat::cmds::net_cmd: attr: content-length,  154381 (services\shellchat\src\cmds\net_cmd.rs:405)
INFO:shellchat::cmds::net_cmd: found content-length of 154381 (services\shellchat\src\cmds\net_cmd.rs:409)
INFO:shellchat::cmds::net_cmd: heap size: 589824 (services\shellchat\src\cmds\net_cmd.rs:420)
INFO:shellchat::cmds::net_cmd: decoding png starting with: [89, 50, 4e, 47, d, a, 1a, a, 0, 0, 0, d, 49, 48, 44, 52, 0, 0, 1, 39, 0, 0, 0, f5, 8, 2, 0, 0, 0, a4, de, 15, dd, 0, 0, 0, 87, 7a, 54, 58, 74, 52, 61, 77, 20, 70, 72, 6f, 66, 69, 6c, 65, 20, 74, 79, 70, 65, 20, 65, 78, 69, 66, 0, 0] (services\shellchat\src\cmds\net_cmd.rs:421)
INFO:shellchat::cmds::net_cmd: heap size: 589824 (services\shellchat\src\cmds\net_cmd.rs:423)
INFO:shellchat::cmds::net_cmd: heap size: 917504 (services\shellchat\src\cmds\net_cmd.rs:427)
INFO:shellchat::cmds::net_cmd: image is 313 x 245 (services\shellchat\src\cmds\net_cmd.rs:430)
```

I suspect what may have happened here is some sort of a race condition on boot, that messed with something inside the kernel because of all the stuff happening at early boot.

I'll go ahead and merge the branch into main, but try to time the heap increase to some "quiet" time to avoid the race condition issue for now...

xobs, 2022-06-27:
That's great that it works!

Is there a way I can reproduce the crash? Or was there any information on the kernel log that indicated what went wrong?

bunnie, 2022-06-27:
Well of course, when I went to go and reproduce the issue by moving the call back to the offending location, it doesn't crash. I'll close this issue for now then, and if I get a case where I see it I'll make a fork and commit that and open a new issue pointing to it.

closed 2022-06-27 by bunnie

bunnie, 2022-06-28:
You know, I encountered the problem of what I thought was the race condition on boot, but actually it was triggered by something else. I think the actual problem is that the PDDB was skating very close to its heap allocation limit, and adding that code to the top broke it.

So I think that was a red herring -- probably this routine is just fine as is!