betrusted-io/xous-core #210: fix `no_delay` setting in `net` (nagle's algorithm state not reported correctly)

bunnie, 2022-08-04:
There's a small bug in smoltcp that was fixed in https://github.com/smoltcp-rs/smoltcp/commit/8090469c2a1456b87a3e277e5b2f2e34e4f7555b

Which prevents us from correctly reporting the state of nagle's algorithm. 

This issue helps anyone who encounters this to know what's up, and to remind me to fix this once they tag out a new release.

dumblob, 2023-06-01:
They seem to have released it in Feb 2023.

bunnie, 2023-07-05:
I did a quick assessment of the changes smoltcp has gone through since 0.8.1 (the version we're pinned at) and the version that supports this (0.9.2+). 

It's...very very substantial. Almost a total rewrite of the library. This is not a day project to pull this in...probably more like a 2-4 week project to refactor the stack.

bunnie, 2023-10-03:
Fixed in #419

closed 2023-10-03 by bunnie