betrusted-io/xous-core #211: Feature: change unlock PIN

duncancmt, 2022-08-09:
Currently, there is no way to change the unlock PIN once it's set. Unlike the root passphrase, due to the semi-frequent requirement to enter it, rotating the unlock PIN seems wise if the device owner suspects compromise (e.g. shoulder surfing while entering the unlock PIN in a public place; forced to hand over the unlock PIN by relevant authorities while crossing a border).

bunnie, 2022-08-09:
Thanks, that's a pretty sound argument for the feature.

bunnie, 2022-08-12:
Alright, this just got implemented and it should be in the "bleeding edge" branch right now.

The PIN change command is buried in the PDDB submenu. It should work "about how you think it should" once you start it. It's pretty much zero-cost to change the PIN because we just shift the wrapped key stored on disk. It's about the only thing that is that easy to shift in the whole system -- most other keys require re-encrypting gobs of data.

As a side note, during first boot the boot PIN is no longer requested, since it's not used at that point (it's used by the PDDB). So you only have to enter it once now, when you initialize the PDDB.

closed 2022-08-12 by bunnie

duncancmt, 2022-09-04:
This feature works on my device :+1: