betrusted-io/xous-core #218: vault: feature request: support for HOTP

kop316, 2022-09-14:
Hello!

The vault app only supports TOTP, however some authentication mechanisms (DUO Mobile) only support HOTP. It would be nice if vault could additionally support HOTP.

kop316, 2022-09-14:
> Can you drop a sample of a (fake, obviously) export record from the HOTP tool you've written? It'll be helpful to have (ideally a couple) of test cases.

```
otpauth://hotp/kop316?secret=REDACTEDSECRET&issuer=Duo&counter=0
```

where `kop316` is the name, `secret` is obviously the secret in base32 (I think?), `issuer` is who gave me it, and `counter` is the counter on it.

> 
> UX integration would be...weird. There's only slots for three apps, because of the number of Fn keys we have. I'd want to glom this into an existing UX - I'm thinking it shows up on the TOTP list, but, the number doesn't update until you select it and tell it to increment the HOTP.

This is how other 2FA apps do it as well, they show up on the TOTP list, and when you click on it, the counter increments and the secret changes.

bunnie, 2022-09-14:
cool thanks. I may not be able to get to this immediately, but I think it's doable and is on the list of things to eventually get done.

bunnie, 2022-09-16:
are you able to build from source and flash to your device? if so, could you please test this branch:

https://github.com/betrusted-io/xous-core/tree/hotp

This adds hotp support in the device. When you create a TOTP record you can specify if TOTP or HOTP; it will default to a count of 0 for HOTP.

When you edit the record, you'll see a field where you can specify totp or hotp (last field). In the case of hotp, the step_by field is now your count field, and it'll increment by 1 automatically every time you use the "autotype" feature.

Would be curious if the logic to generate the HOTP even works though -- from my reading of the spec it's identical to TOTP except that you're just passing the count instead of the time divided by the seconds step.

Note that the backup/restore host code has not been updated to support HOTP. Would like to first check that HOTP even works on the device first. Also, I didn't write the backup/restore tool so it's a lot harder for me to modify it to do a new thing.

kop316, 2022-09-17:
Hello!

I imagine I can figure out how to compile and flash it. Is it ok if this doesn't get done until Sunday or Monday?

bunnie, 2022-09-17:
yah, should be fine. thanks!

kop316, 2022-09-19:
I can confirm HOTP worked correctly!

One UI request, when entering HOTP, could the app prompt for the count number? That way if it is a backup from another app, I can just input it there vs having to manually edit it.

kop316, 2022-09-19:
Ok, well after rebooting, when I go to the vault app and attempt to go to the `TOTP` I get the error:

[image]

I only have the HOTP entry

bunnie, 2022-09-19:
ah hm. i have an inkling of what might be going on. let me poke at it a bit....i'm reusing the step_by field and i bet I'm not isolating a case correctly in the hotp code so with a count = 0 you end up with a divide-by-zero error

bunnie, 2022-09-19:
I just pushed 22a440f75f12f2d06fb97da58bd2e1f0b1ba2469 to main (HOTP is in main now) that adds your requested initial count feature, and should at the very least turn the panic into a warning printed on the console; but ideally, it fixes the issue entirely.

kop316, 2022-09-20:
This is working, so I am closing. Thank you!

closed 2022-09-20 by kop316