betrusted-io/xous-core #223: Vault: provide a UI action to cancel a webauthn request

kop316, 2022-09-21:
Currently, if the vault app receives a webauthn request, pressing any key confirms the request and the vault app responds. The only way to cancel a request is to disconnect the USB cable.

It would be nice if there was a way to cancel/ignore the request, e.g. for a malicious request, or a malfunctioning device.

bunnie, 2022-09-21:
I *thought* there is a 9 second timeout or something for webauthn requests -- it should just timeout and deny the request at the end of that (and there is a little countdown timer). This is part of the FIDO2 spec and it's in the testbench so that works.

However, maybe there is a different code path. In particular for the U2F (older FIDO standard) the UI for that is a bit of a hack and maybe that's what you're referring to.

The problem is without knowing exactly what mode you're seeing this in, there's too many potential paths where this problem could be. I know you probably don't want to share a photo of the credential since it has some sensitive infos, so, I'd suggest one of two things:

1. Do a debug build and try to capture the serial logs where this happens (would require you to use the pi hat and plug in the dev cable etc.)
2. Go to `demo.yubico.com` and find a flow that triggers the behavior you're seeing, and send me the instructions so I can try to reproduce.

kop316, 2022-09-21:
> I _thought_ there is a 9 second timeout or something for webauthn requests -- it should just timeout and deny the request at the end of that (and there is a little countdown timer). This is part of the FIDO2 spec and it's in the testbench so that works.

Is the timeout a client side timeout or a server timeout? I tested the webauthn on both github.com and `demo.yubikey.com` as you suggested, and the UI window stays up until the server times out (it looked to be ~20-30 seconds?). As I am showing in the image below, I do not see a timeout.

> However, maybe there is a different code path. In particular for the U2F (older FIDO standard) the UI for that is a bit of a hack and maybe that's what you're referring to.
>
> The problem is without [...]

Here is am image: 
[image]

Hopefully that makes it clear!

I know you probably don't want to share a photo of the credential since it has some sensitive infos, so, I'd suggest one of two things:
> 
>     1. Do a debug build and try to capture the serial logs where this happens (would require you to use the pi hat and plug in the dev cable etc.)
> 
>     2. Go to `demo.yubico.com` and find a flow that triggers the behavior you're seeing, and send me the instructions so I can try to reproduce.

I am able to recreate the issue if I stay in the `FIDO` window use `demo.yubico.com` in the default settings.

bunnie, 2022-09-21:
OK yep this is the older U2F flow, which is a bit of a hack. I think I know where that might need to be fixed for the timeout, I'll have a think about it later.

bunnie, 2022-09-22:
Alright, after reading a bit on the specs, here's what I think is the case:

1. You actually can't decline the presence test, according to the spec. This is because, well, on normal U2F devices, you get one button and a light. There isn't a way to actually say no.
2. So, in theory, to be spec-compliant, the behavior you described is "as designed".
3. There might be a non-spec compliant way to send an error code to the browser if you decline things, but, if we were to do this, it could cause unpredictable things to happen. Browsers aren't programmed to handle the U2F device throwing an error in this path, so it could make things "worse".

My inclination is to mark this as "as designed", rather than try to extend the spec because doing so would probably cause a series of subordinate bugs to be opened against different browsers and versions of browsers, each with different and unique undefined behavior, as we hit edge case code that is normally impossible to hit.

Thoughts?

kop316, 2022-09-22:
My thought for it is, instead of sending an error code, it is more of an "ignore". So on the UI side, the prompt to respond goes away (to allow the user to go about whatever they were doing), and under the hood, it is just as if the user never clicks on the button to acknowledge the request.

I imagine that would be spec compliant, because a browser has to be able to handle a device never responding to the request?

bunnie, 2022-09-22:
Hm, that's an interesting idea. The problem is that the host is still actually constantly spamming the device with keepalive packets during this period to poll for an interaction. If we dismiss the UI state but the host is still talking to the device, the user would be free to do something else that could interfere with the USB protocol. 

I suppose maybe an intermediate solution could be to add a message with something like "Press F4 to ignore" and then it'll keep up a box saying "waiting for host to cancel request..." in this case you'd be spec compliant but the user could have the satisfaction of taking positive action to reject the authorization request?

kop316, 2022-09-22:
> Hm, that's an interesting idea. The problem is that the host is still actually constantly spamming the device with keepalive packets during this period to poll for an interaction. If we dismiss the [...]

Ahh, that makes sense. 

> I suppose maybe an intermediate solution could be to add a message with something like "Press F4 to ignore" and then it'll keep up a box saying "waiting for host to cancel request..." in this case [...]

I think this is a good compromise! Also, that way, the user knows that the host has to dismiss the request (or they need to disconnect USB) before the UI will go away, and the user won't be worried about accidentally authorizing the request as well.

bunnie, 2022-09-22:
I just pushed a patch that adds the described feature. I think it's OK? give it a spin from ci and let me know what you think.

kop316, 2022-09-22:
I think that is a good UI flow, I can close this then. Thanks!

closed 2022-09-22 by kop316