betrusted-io/xous-core #224: Vault App: Guru Meditation panic when deleting a FIDO entry

kop316, 2022-09-21:
In working on #223 , I added a demo key from `demo.yubikey.com`. However when I go to delete it, I get the following:

[image]

[image]

[image]

The device resets after ~20 seconds. Once I go into the vault app again, the demo key is still there, so I am unable to delete it.

bunnie, 2022-09-21:
Thanks for reporting this. will have a look in a couple of days.

bunnie, 2022-09-22:
Just pushed a fix to main that should resolve this one. thanks for reporting it!

commit 13402b99 (2022-09-22): fix #224

was a logic bug where the database name was not updated while
doing some refactoring previously
closed 2022-09-22 by bunnie