betrusted-io/xous-core #225: Xous Loader seems to hang on `Loading Kernel`

kop316, 2022-09-23:
About ~50% of the time I try to wake from sleep or reboot, the Xous loader seems to hang on `Loading Kernel`. I usually wait ~20-30 seconds and get impatient and push the paperclip button. However, I seem to have to push the paperclip button at least 2 times to get out of this hanging loop.

[image]

bunnie, 2022-09-23:
Hanging on the loader is usually indicative of a failure to resume after suspend. I could see https://github.com/betrusted-io/xous-core/commit/6cf5135a7d833862c0c53d63d5e2e12e08a6dcac having something in it that mucks with the resume path -- i stripped out an unused interrupt, I may have forgotten to mask something out that's causing spurious events.

I'll have a look at it later tonight. You said it happened before this commit -- was it very repeatably happening? What sorts of things were you doing just before putting the system into sleep?

suspend/resume failures are a bit tricky to trace down without a serial console log, so I'll probably need some help to reproduce this.

kop316, 2022-09-23:
> You said it happened before this commit -- was it very repeatably happening? 

The earliest commit I can say it started on was https://github.com/betrusted-io/xous-core/commit/22a440f75f12f2d06fb97da58bd2e1f0b1ba2469 (since I never really used the Precursor beforehand)

> What sorts of things were you doing just before putting the system into sleep?

I really only use the Precursor for the Vault app, so maybe there is an interaction with the Vault app? However, there does not seem to be an interaction that I can reliably cause this issue. 

> suspend/resume failures are a bit tricky to trace down without a serial console log, so I'll probably need some help to reproduce this.

If there is something I can do to help, please let me know! I do not have a rasp-pi, so I don't know how much access I have to the serial interface.

bunnie, 2022-09-23:
Alright, I just committed d3daca9c which might help the issue. This is what I found:

1. I did a suspend/resume stress test. On the third round, the time server did not suspend correctly due to contention for the llio mechanism. However, the suspend is a "null" op; what's more important is that time is notified of a resume so that it can resync time on resume.
2. I rewrote the time server to use the existing private for secure ops for the suspend/resume request. This would reduce the chance of the request failing to nil
3. I ran the stress test and it completed 143 cycles successfully, at which point I stopped testing

Maybe this fixes it, maybe it doesn't -- that's the nature of suspend/resume bugs. I just reset the CI system to kick off a new build, so you should be able to pick it up in a few minutes.

kop316, 2022-09-23:
I just installed, I will let you know what I find!

kop316, 2022-10-01:
I haven't encountered this in a week, so I am closing. Thank you!

closed 2022-10-01 by kop316