betrusted-io/xous-core #269: PDDB basis opening ordering display

eau-u4f, 2022-11-04:
as referenced in this issue (#194 ) conversation with @bunnie and my misunderstanding of basis:
the basis are behaving like a stack, the writing will be done on the basis residing on top of the "opened basis" stack,
it would make it clearer to have a hint on which is the currently "top-of-the-stack" basis, before editing or writing entries.
- either bold/or a '*' next to the entry in the "manage bases" menu.
- in the status bar "basisA > basisB > basisC" (but can be long if you have 3+ basis).
- just display the last opened basis in the status bar or the last 2 overlays.
- something else?

Is there somewhere I could start documenting this to avoid further misunderstanding, of course it might not be as clear as the creator, but could be a start? shall I write quickly in a .txt file?

bunnie, 2022-12-28:
2f30e1b50a927e622d4a504c5cf8712ce09431d2 and 5370c333da7c9f04e03e029bda35e05eb65ed09c implement this feature. I think it's good enough...if you have many Bases open with long names, the names will go off the edge of the screen (or will it word wrap?...let me know if you find out :grin: ), but I think it's OK because it's listed left-to-right with the basis that has priority, so the ones that gets cut off on the right isn't so relevant anyways.

[image]

[image]

bunnie, 2022-12-28:
I'll close the issue out for now because I think it's implemented as much as it could be, but feel free to open it if there's other requests you'd like to make. Sorry it took a while, holiday duties kinda got in the way of work for a bit.

closed 2022-12-28 by bunnie

eau-u4f, 2023-05-28:
no yep, i notice it's now written on top, i won't hesitate if I have other comments.
thanks A LOT, for your product, commitment and help.