betrusted-io/xous-core #276: cargo xtask renode-image error

eau-u4f, 2022-11-13:
Trying to prepare renode, and error when building renode-image.
From:
https://github.com/betrusted-io/xous-core#quickstart-using-an-emulator
```
$ cargo xtask renode-image
    Updating git repository `https://github.com/betrusted-io/curve25519-dalek.git`
    Updating git repository `https://github.com/betrusted-io/ring-xous`
    Updating git repository `https://github.com/betrusted-io/usb-device.git`
    Updating crates.io index
    Updating git repository `https://github.com/betrusted-io/com_rs.git`
    Updating git repository `https://github.com/betrusted-io/engine25519-as.git`
  Downloaded serde v1.0.139
  Downloaded unicode-ident v1.0.0
  Downloaded cc v1.0.73
  Downloaded bzip2 v0.4.3
  Downloaded ureq v2.5.0
  Downloaded bzip2-sys v0.1.11+1.0.8
  Downloaded fastrand v1.7.0
  Downloaded itoa v1.0.2
  Downloaded num-traits v0.2.15
  Downloaded num-integer v0.1.45
  Downloaded memchr v2.5.0
  Downloaded form_urlencoded v1.0.1
  Downloaded filetime v0.2.17
  Downloaded chunked_transfer v1.4.0
  Downloaded once_cell v1.12.0
  Downloaded rustc_version v0.4.0
  Downloaded quick-xml v0.19.0
  Downloaded ryu v1.0.10
  Downloaded remove_dir_all v0.5.3
  Downloaded percent-encoding v2.2.0
  Downloaded matches v0.1.9
  Downloaded proc-macro2 v1.0.39
  Downloaded time v0.1.44
  Downloaded semver v1.0.9
  Downloaded quote v1.0.18
  Downloaded serde_derive v1.0.139
  Downloaded serde_json v1.0.82
  Downloaded thiserror v1.0.32
  Downloaded syn v1.0.95
  Downloaded thiserror-impl v1.0.32
  Downloaded webpki-roots v0.22.4
  Downloaded sct v0.7.0
  Downloaded untrusted v0.7.1
  Downloaded pkg-config v0.3.25
  Downloaded miniz_oxide v0.5.1
  Downloaded tinyvec_macros v0.1.0
  Downloaded zip v0.5.13
  Downloaded tinyvec v1.6.0
  Downloaded unicode-bidi v0.3.8
  Downloaded tempfile v3.3.0
  Downloaded webpki v0.22.0
  Downloaded unicode-normalization v0.1.21
  Downloaded rustls v0.20.6
  Downloaded url v2.2.2
  Downloaded cfg-if v1.0.0
  Downloaded adler v1.0.2
  Downloaded byteorder v1.4.3
  Downloaded log v0.4.17
  Downloaded chrono v0.4.19
  Downloaded flate2 v1.0.24
  Downloaded crc32fast v1.3.2
  Downloaded idna v0.2.3
  Downloaded convert_case v0.4.0
  Downloaded base64 v0.13.0
  Downloaded autocfg v1.1.0
  Downloaded atty v0.2.14
  Downloaded libc v0.2.126
  Downloaded svd2utra v0.1.7
  Downloaded 58 crates (4.3 MB) in 2.21s
[... 269 lines cut ...]
7 | #[derive(Clone, Copy)]
  |   ^^^^^^

error: cannot find macro `debug_assert` in this scope
  --> /Users/eau/.cargo/registry/src/github.com-1ecc6299db9ec823/seahash-4.1.0/src/reference.rs:38:5
   |
38 |     debug_assert!(
   |     ^^^^^^^^^^^^

error: cannot find attribute `derive` in this scope
 --> /Users/eau/.cargo/registry/src/github.com-1ecc6299db9ec823/seahash-4.1.0/src/buffer.rs:8:3
  |
8 | #[derive(Clone)]
  |   ^^^^^^

error[E0405]: cannot find trait `Default` in this scope
  --> /Users/eau/.cargo/registry/src/github.com-1ecc6299db9ec823/seahash-4.1.0/src/stream.rs:19:6
   |
19 | impl Default for SeaHasher {
   |      ^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0405`.
error: could not compile `seahash` due to 7 previous errors
Error: "Local build failed"
```

xobs, 2022-11-14:
That's bizarre. It's complaining that you're compiling with a different version of rustc. It's saying you're using the 1.64.0 toolchain, and it... really wants the 1.64.0 toolchain.

Can you please provide the output of `rustc --version`?

eau-u4f, 2022-11-15:
$ rustc --version
rustc 1.64.0

xobs, 2022-11-15:
Are you using a custom build of rustc? If you use the official version, it prints out the git hash that built it:
```sh
$ $ rustc --version
rustc 1.64.0 (a55dd71d5 2022-09-19)
$
```

If you use a custom build of Rust, you will need to build libstd for Xous yourself. Instructions on how to do that are made available at: https://github.com/betrusted-io/rust#building-from-source

eau-u4f, 2022-11-15:
homebrew version on macOS

eau-u4f, 2022-11-17:
i'll try the build from source.

xobs, 2022-11-17:
It looks like homebrew ships a custom build of Rust, and is therefore incompatible with the precompiled libstd we ship: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/rust.rb

Since that's the case, you'll need to build libstd from source.

bunnie, 2022-12-02:
Should this issue be marked as wontfix and closed because the resolution seems to be that an incompatible toolchain build was used?

xobs, 2023-02-02:
In the future, building your own libstd will be slightly easier when using nightly and `-Zbuild-std`, but for now it's a "wontfix" -- we only support official Rust builds.

closed 2023-02-02 by xobs