betrusted-io/xous-core #281: Vault App: Editing a password erases entire sections instead of editing it

leonlup0, 2022-11-25:
## Steps to reproduce:

1. Open Vault app, press F3 to open password manager
2. Press F4 and "Add new item"
3. Type domain (i.e. `example.c`)
4. Type username (i.e. `foo`)
5. Proceed with the suggested password
6. Scroll to the new added password, press F4 and select "Edit item"
7. Try fixing the domain so it reads correctly (i.e. `example.com`)

**Problem:** Any keypress will delete the contents of the line selected and replace with whatever key was pressed. This happens in all lines.

**Expected behaviour:** keypresses are appended to the end of the line. This is especially useful when trying to append a symbol to the end of a previously generated password that has no symbols. Or when appending more information to the notes area.

leonlup0, 2022-11-25:
Another thing that may be another issue, there is currently no way of viewing additional notes written to a password entry other than editing it. This can lead to further frustration, one wrong keypress and you may accidentally delete a password or loose the data written in notes.

bunnie, 2022-11-25:
That's good feedback. the way ti's implemented is that the text in the fields uses the "suggested default" mechanism (so it's treated as a placeholder). Not obviously, the way to use the placeholder is to push the D-pad keys left or right to "keep" the placeholder, before typing.

I think you are correct in that a less surprising behavior would be if the fields persisted even if you started typing. This does mean that the initial word "notes" for example would stick around in the field and you'd have to backspace over it to remove that, and the initial password generated if any would need to be backspaced over to delete it. 

But I think that is probably a minor sacrifice in exchange for a more intuitive behavior for passowrds that have been set up and durably stored in the device.

implementing this request will require adding another input method style to a core modal mechanism, so I will need to add this to the task list, it's not a one-liner to configure something and push the result. But I think we should do this, your issue would be a common one and I agree the behavior is surprising and not desireable.

leonlup0, 2022-11-25:
This is related, but probably should be separated in another issue:

Maybe adding a "cancel" button would also be beneficial. There are times when you do an action, like, "Set Time" and you want to cancel, and you can't really, you need to go through the whole process of resetting your time. If I do an edit on a password but I don't want to save it, a "Cancel" button would be beneficial/expected.

bunnie, 2022-11-26:
Yah, this would be a separate issue. Cancel button logic is surprisingly hard because each instance of a cancel button adds another branch path on a UX flow that has to be handled -- a flow that is otherwise linear in many cases. You're not wrong that some flows could benefit from it, but also, it does add a bit of complexity to the logic, because the thing requesting the dialog box now has to handle falliable operations. That being said some things like time setting could be cancelable without much issue, but it would also be something to consider on a case by case basis.

bunnie, 2022-12-02:
alright, this should be fixed now in the bleeding edge.

closed 2022-12-02 by bunnie