betrusted-io/xous-core #286: pddb dict cannot be created by mtxcli on device (but can be added to)

plaurent, 2022-12-26:
# Overview

The Matrix messenger `mtxcli` Xous app uses `pddb` to store matrix login credentials. On device, when following the instructions for setting user and password, an error occurs and the user cannot proceed without a workaround outside of the `mtxcli` app.

# Steps to reproduce

* Install mtxcli onto precursor via app image
  * `cargo xtask app-image mtxcli`
  * `python3 ./tools/usb_update.py -k ./target/riscv32imac-unknown-xous-elf/release/xous.img`
* In `shellchat`, confirm you don't yet have `mtxcli` dict 
  * `pddb dictlist` (output should not contain `mtxcli`)
* Switch to App `mtxcli` using the home menu button in Xous
* Attempt to set username:  
  * `/set user abc`
* Unexpected result: 
   * The set fails fails; a circle is printed. 
   * `/get user` responds with `user is UNSET`. (See screenshot.)
   * Also, in `shellchat` you can see in `pddb dictlist` that you still don't have an `mtxcli` dict.

Here is the log showing the apparent `pddb` misbehavior:

```
INFO:mtxcli::cmds: set 'user' = 'abc' (apps/mtxcli/src/cmds.rs:104)
ERR :pddb::libstd: path mtxcli couldn't be found (services/pddb/src/libstd/mod.rs:129)
INFO:mtxcli::cmds: dict 'mtxcli' exists (apps/mtxcli/src/cmds.rs:108)
ERR :pddb::libstd: unable to find key user in dict mtxcli (services/pddb/src/libstd/mod.rs:487)
ERR :mtxcli::cmds::set: error setting key user: Custom { kind: Other, error: "error occurred when opening file" } (apps/mtxcli/src/cmds/set.rs:41)
```

# Workaround

* Switch back to `shellchat` and use `pddb write mtxcli:user abc` to create the dictionary. After having done that, subsequent writes from mtxcli (like /set password) work.

[image]

plaurent, 2022-12-27:
In the Xous App Development chat, @xobs suggested a possible underlying source for the issue: "Now that libstd fs support has a Real User, it seems like we're finding Bugs... "
https://matrix.to/#/!aNXzIBexYHdciytSVg:matrix.org/$GYfTdqXGQARY2RnHDSznIvoXziBsMD3HcPwKatGxQfw?via=matrix.org&via=cofree.coffee&via=pussy.accountants

bunnie, 2022-12-27:
Confirmed this is a bug in `std`. I have submitted

https://github.com/betrusted-io/rust/pull/3

Once @xobs reviews and tags it a new version of the toolchain will get built which resolves this issue.

bunnie, 2022-12-27:
This is now pushed and available as https://github.com/betrusted-io/rust/releases/tag/1.66.0.2

You will need to re-fetch your toolchain to build a version that doesn't have this bug, or pull from our CI system.

closed 2022-12-27 by bunnie

plaurent, 2022-12-28:
Confirmed, the initial `/set user`, now works from within `mtxcli` on the device. Thank you @bunnie and @xobs !