betrusted-io/xous-core #287: mtxcli on device crashes with guru meditation

plaurent, 2022-12-26:
# Overview

On device, the Matrix messenger `mtxcli` Xous app can successfully connect to a server/room, but upon receiving messages from the room, crashes.  However, `mtxcli` works as expected, without crashing, in hosted mode on a computer.

@bunnie has proposed "if you have steps to reproduce then I can try to add the heap expansion code and see if that fixes it"

# Steps to reproduce

* Install mtxcli onto precursor via app image
  * `cargo xtask app-image mtxcli`
  * `python3 ./tools/usb_update.py -k ./target/riscv32imac-unknown-xous-elf/release/xous.img`
* Switch to App `mtxcli` using the home menu button in Xous
* Configure `mtxcli` witwh your Matrix account as per the README: https://github.com/betrusted-io/xous-core/tree/main/apps/mtxcli#setup
  * If the setup instructions fail, use the Workaround here: https://github.com/betrusted-io/xous-core/issues/286
  * You could join a test room I created, `#mtxcli1:matrix.org`  (e.g., `/user @[username]:matrix.org`, `/password [password]`, `/room #mtxcli1`).
* Type `/login` to connect
  * If connecting fails with authentication errors, try switching back to `shellchat` and executing `net tcpget [some_url/item]`. Then switch back to `mtxcli` and try again.
* Optionally type a message.
* Press enter (this will attempt to retrieve messages from the room on the server).
* Unexpected result:
  * a modal appears with error title `~~Guru Meditation~~` and message `memory allocation of 408 bytes failed`

The device is frozen/crashed after this point.

# Workaround

No known workaround -- the app is not currently usable for reading room messages on device.

  
[image]

bunnie, 2022-12-27:
Hm, i think the error might cut deeper than just a simple oom. I've instrumented the client and net stack to print out the heap usage as it's running, and I don't see mtxcli app going above 256k, and the net stack stays below 64k.

I can't reliably reproduce the OOM, anyways...more often than not I just get a bunch of messages about Rx packet wakeups being punted into the future, which is just the net stack debug informing that the smoltcp async wakeup algorithm is requesting too many wakeups (I think it's basically harmless, since a wakeup *is* scheduled and at the next wakeup it'll handle everything, but I haven't figured out why smoltcp does this in the first place, and if you don't limit the wakeup requests it can make it'll exhaust resources).

Matrix is also sending me varying results when I try to connect, so reproducibility is low. Currently, I'm getting this error:
```
ERR :mtxcli::cmds::web: ERROR code 403 err_body = {"errcode":"M_FORBIDDEN","error":"Unknown room"} (apps\mtxcli\src\cmds\web.rs:57)
ERR :mtxcli::cmds::web: Error for send_message (apps\mtxcli\src\cmds\web.rs:442
```

Should the room be `#mtxcli1`, or `#mtxcli1:matrix.org`?

Above you indicated that the command should be `/room #mtxcli1`, but in fact, the full command should be `/set room #mtxcli1`? or should it be `/set room #mtxcli1:matrix.org`?

plaurent, 2022-12-27:
Apologies for the confusion, the correct command is `/set room #mtxcli1`

(https://github.com/betrusted-io/xous-core/tree/main/apps/mtxcli#setup)

bunnie, 2022-12-27:
thanks. I do see the panic, and tracing along with the kernel in real time as it happens, it looks like the code is actually allocating megabytes of memory rapidly, so there is either a bug of some sort in the client itself, or the server is trying to send some large amount of data that this client can't handle. 

I might need to refer this to @tmarble since he wrote the client code. Tom, the client seems to be able to get itself into a state where it's talking to the server and just receiving tons of packets, but I don't have any visibility into what it is or why. Do you have any ideas?

tmarble, 2022-12-27:
Yes, I have a good sense for this...
There are two optimizations which have not yet been made: narrowing client
sync to only on room and only certain mutations.

If a user is following many, active rooms the sync might result in a
cascade of updates.

I will work on this.

--Tom

bunnie, 2022-12-28:
The description matches the behavior...I'm following a bunch of rooms, so, if it is grabbing the conversation history of all the rooms I'm following, that would be quite a bit of data to grab.

Thanks for looking into this!

bunnie, 2022-12-28:
I just pushed 8d42feb0514b634389d1a5b7fe17e92582aaa408

This increases the stack & heap space to 1MiB and 2MiB limits, respectively, and I've added a `heap_usage()` call so you can get some feedback on how memory usage is going. 

If you need more heap we can go beyond 2MiB but generally if you're just storing text data and your chat history isn't more than a couple thousand lines deep, you won't need much more storage than that.

plaurent, 2022-12-29:
To add supporting evidence for the the above hypothesis (that the crash is about grabbing lots of conversation history data), I created a second Matrix account and subscribed to just 1 quiet room. I found that with that new account, `mtxcli` worked fine on the device and did not crash.

bunnie, 2023-01-18:
@tmarble's most recent PR seems to fix this issue. Would be curious if it helps for you @plaurent .

bunnie, 2023-01-18:
(fwiw I have merged his PR so it will be available on bleeding-edge in about 10 minutes assuming CI goes well)

plaurent, 2023-01-19:
Yes!!! This fixed it!! Thank you @tmarble!! Closing this issue!

closed 2023-01-19 by plaurent