betrusted-io/xous-core #297: pddb write shorter value doesn't truncate longer value

plaurent, 2023-01-14:
Using `pddb write` for a value shorter than what's already stored under the same key doesn't chop off the value's old bytes.  For example writing a value `hello` over `goodbye` results in `helloye`.  Is this the expected behavior?  Is the only way to have a shorter value to first delete the old key/value?

Example screenshot:

[image]

bunnie, 2023-01-15:
The semantics were designed to match this:

https://doc.rust-lang.org/std/fs/struct.OpenOptions.html#method.write

If you use this call to open a file and write something less than the amount of data in the buffer, you get the new stuff + remainder of old stuff.

Strangely enough, in Rust, `std::fs::Open` can only open files for reading. If you want to open a file for writing, it offers one call, which is `Create:

https://doc.rust-lang.org/std/fs/struct.File.html#method.create

The `Create` method has the semantics you're expecting.

It might be the case that since we have `std` available, the `Create` method may have an auto-implementation that does what you're expecting.

I'd say my response is more just a rationale for why things are what they are. However, it doesn't mean I made the right choice. I'm open to hearing arguments for why we should deviate from Rust semantics (for example, Python `write` does truncate, but Rust `write` doesn't, and I don't know why one or the other would be chosen), and/or if maybe we should rename the PDDB shell command to be more intuitive, but perhaps leave the low-level API as-is.

plaurent, 2023-01-15:
To me “write” does indeed sound like something that lives in a filesystem’s semantic space — and I can see how it ought to follow the defaults for the implementing language. I’d think a “set” — if one existed — would be more in keeping with a dictionary’s semantics, and ought to exhibit the behavior that I would expect.

However — I would be curious if there are any pddb use cases where the default “write” behavior would be preferred over a “set”?  

I suppose this also has to do with abstraction… should pddb really offer dictionary-like semantics, abstracting beyond Rust’s std filesystem’s?

bunnie, 2023-01-18:
Yah, I really don't know the answer to this. But, I think there is value in both APIs (overwrite, and truncate). 

My gut is to go with Rust semantics, even though this is a k/v store...so `write()` will just stick bytes in an existing file, leaving the remainder, and some sort of `create`-style API (which doesn't exist) would write data and trucate any existing data once closed.

This is different from how things should behave at the `shellchat` level -- you're probably correct in that a user command of `write` might very well be least surprising if writing to the same-named file truncated the excess data.

xobs, 2023-01-18:
As a matter of trivia, the `.truncate()` function does exist in Rust as part of `OpenOptions`, though I haven't tested it out: https://doc.rust-lang.org/std/fs/struct.OpenOptions.html#method.truncate

bunnie, 2023-07-05:
#352 fixes this, so, closing this issue.

closed 2023-07-05 by bunnie