betrusted-io/xous-core #305: rtc: time calls block until pddb is running

xobs, 2023-01-29:
As part of libstd automated tests, I'm running through most of the OS functionality.

Right now it's getting caught at the RTC tests. Specifically, `time::tests::since_epoch`. This makes a call to the time server using `GetUtcTimeMs`. However, the time server seems to start a dummy server early on but then blocks waiting for the PDDB (including userprefs).

The filesystem appears to be used for adding localtime TZ offsets as well as a UTC offset (?). 

The `GetUtcTimeMs` call shouldn't even worry about the timezone, so that shouldn't be blocked by the filesystem being unavailable.

bunnie, 2023-01-29:
The PDDB stores the actual offset between what the RTC reports and "reality".

It turns out that the RTC hardware itself is flawed -- if you try to use it as an actual RTC, it doesn't track leap years, among other things. It also only tracks years up to 99 years, so dates from 1999 are indistinguishable from the year 2099 (e.g. it has much less resolution than 32 bits). 

In order to work around this, the RTC hardware is instead used simply as a "seconds since initialized" counter -- we ignore anything it tries to report about days/months/weeks/years and simply try to extract as many bits as we can out of it as a simple seconds counter.

This gives us the ability to track time for roughly 100 years since the initialization point before we suffer a rollover, instead of an increasingly short amount of time as initialization dates start to approach the end of a century (I know, largely theoretical at this point since we're in the year 2023, but still feels awful to have coded into the US something that's guaranteed to break one year after a system is initialized in 2099).

The trade-off is you need to know how long it's been since Unix Epoch to whenever the RTC was initialized, because the RTC merely tracks seconds elapsed since initialization, and not an absolute date. That's the value coming out of the PDDB, and that is why you can't do a `GetUtcTimeMs` call until you can know that offset.

bunnie, 2023-01-30:
Alright, so, you want the following behavior:

- `GetUtcTimeMs` always returns
- It may return a totally bogus value
- The bogus value is allowed to be arbitrarily far in the future or far in the past relative to true time
- Can the bogus value could be the same value all the time, or a completely random value? Or do you expect the bogus value to still behave timer-like? The easiest thing to do if bogus can truly be "anything" is to just return a constant number. Otherwise maybe we could just return the value of the ticktimer, which would "generally set you near epoch" on boot.
- When the PDDB is mounted, the value will abruptly change to the true value. This change will be by a random and large amount without any warning.
- After this point, the RTC behaves as a linear, monotonic source of time
- The API would require using the new `scalar5` return mechanism, since the result should be a `u64` and I'm loathe to chop a bit off of the returned value to indicate accurate or made-up?

xobs, 2023-01-30:
The bogus value can be constant or completely random. Nobody should be using this value for actual timekeeping. This value will be used for things like file timestamps or TLS negotiations, but since those won't happen without PDDB (or some other mechanism of correcting the time) it's fine.

`scalar5` is fine to use.

bunnie, 2023-01-30:
I...actually don't see this behavior. The server doesn't block on the PDDB mounting. I wrote this small test routine:

```rust
                    let time_cid = xous::connect(xous::SID::from_bytes(b"timeserverpublic").unwrap()).unwrap();
                    let result = xous::send_message(time_cid,
                        xous::Message::new_blocking_scalar(3, 0, 0, 0, 0)
                    ).unwrap();
                    match result {
                        xous::Result::Scalar2(msb, lsb) => {
                            log::info!("GetTimeUtc: {}, {}", msb, lsb);
                        }
                        _ => {
                            log::info!("GetTimeUtc returned an unexpected result");
                        }
                    }
```

And it will complete whether or not the PDDB is mounted.

The routine actually does return a value, it's just the base hardware time counter result without the UTC offset....

There's a couple of options for why you may be seeing it block in your test environment though:

1. If the RTC isn't returning a "valid" result (there are some invalid time codes for the BCD-coded fields of the hardware chip), the system will repeatedly try to correct that. Could be a Renode emulation issue with the RTC hardware in that case. See below:
```rust
            // on boot, do the validation checks of the RTC. If it is not initialized or corrupted, fix it.
            let mut settings = [0u8; 8];
            loop {
                match i2c.i2c_read(ABRTCMC_I2C_ADR, ABRTCMC_CONTROL3, &mut settings) {
                    Ok(I2cStatus::ResponseReadOk) => break,
                    _ => {
                        log::error!("Couldn't check RTC, retrying!");
                        xous::yield_slice(); // recheck in a fast loop, we really should be able to grab this resource on boot.
                    },
                };
            }
```
2. Could be maybe something with the suspend/resume server not being around or responding. If that's missing from your environment, RTC wants to hook the susres so it can properly manage re-creating the local time when the SoC comes out of cold storage.

Could either of those be causing the problem you're seeing?

xobs, 2023-01-30:
Correct me if I'm wrong, but this is the main loop that handles time:

https://github.com/betrusted-io/xous-core/blob/main/services/status/src/time.rs#L239-L247

Specifically, this requests a connection to the PDDB server:
https://github.com/betrusted-io/xous-core/blob/797bb0adc6bcccb72f5c36514834b9b45f699735/services/status/src/time.rs#L256

That call is here:

https://github.com/betrusted-io/xous-core/blob/797bb0adc6bcccb72f5c36514834b9b45f699735/services/pddb/src/lib.rs#L44-L49

If PDDB is not running, this call will block forever. PDDB currently cannot be used in a test environment, so this call will never finish and time operations will never return.

bunnie, 2023-01-30:
Oh I see what you mean -- not that the PDDB is not just not mounted, it simply *does not exist* in the system.

Hm. I'll have to contemplate how to fix that without breaking systems that do have a PDDB.

bunnie, 2023-01-30:
How are you even invoking this? The crate that the code is in should summarily fail in multiple spectacular ways if there is no PDDB server on the system.

bunnie, 2023-01-30:
It shouldn't even get past this line:
` let prefs = userprefs::Manager::new(); `

That has a call to `pddb::new()` inside it...

xobs, 2023-01-30:
To get the tests working, I commented out everything involving pddb and userprefs, and set the two values that it relies on to `0`. That was enough to get the tests to run and not hang.

bunnie, 2023-01-30:
I think probably the best way to handle this is with a `#cfg[]` flag for testing. There isn't...a reasonable work-around for connecting to servers that don't exist that also hold state critical to the correct function of the system. The alternative would be to write code that never gets executed on a normal system but is linked in and carried around for a testing configuration that is never run.

So I think a `cfg` flag is probably the way to go to solve this situation.

bunnie, 2023-01-30:
Added a PR that I think resolves the issue?

PR #307 (2023-01-31): move pddb dependencies to a feature flag for testing purposes
Maybe this resolves #305?
closed 2023-01-31 by bunnie