betrusted-io/xous-core #308: hosted mode regression with new `libstd` refactors

bunnie, 2023-01-30:
Hosted mode panics immediately now with this error:
```
Panic Details: PanicInfo { payload: Any { .. }, message: Some(index out of bounds: the len is 32 but the index is 65535), location: Location { file: "kernel\\src\\arch\\hosted\\process.rs", line: 287, col: 17 }, can_unwind: true }
```
I'm guessing this is related to some refactoring done to the process record format. @xobs can you please help to update the corresponding hosted mode files to accommodate the changes? Doesn't need to be performant, just needs to work. thanks!

bunnie, 2023-01-30:
This looks like it's related to the `MULTIPLY_DEBRUIJN_BIT_POSITION` counting on `usize` being 32 bits, but it's 64 bits on x86_64. However, fixing `find_next_thread()` doesn't seem to fully fix the problem. I'm guessing there's another trick I'm missing, but hopefully it's pretty obvious to @xobs what it is.

bunnie, 2023-01-30:
Not sure if this is related, but `cargo test` in the `kernel` directory is also currently broken. I suspect it is related.

xobs, 2023-01-31:
PR #309 seems like it should fix this.

bunnie, 2023-01-31:
It's definitely better! However, when I try to run any of the CI testbenches in hosted mode, I get this:

`
Panic Details: PanicInfo { payload: Any { .. }, message: Some(PID 25 was not in a state to wake thread 2: Running(111101110)), location: Location { file: "kernel\\src\\services.rs", line: 691, col: 22 }, can_unwind: true }
`

You can reproduce this by running `cargo xtask pddb-ci`, it should briefly run and then fail fairly quickly with this result.

The failure mode is consistent over multiple runs, but how long it takes to fail varies just a little bit.

xobs, 2023-01-31:
I thought I might have missed removing some `ss.ready_thread()` calls from various other syscalls. Thanks for finding that.

When I run `cargo xtask pddb-ci` it pops up a window and offers to reformat the image. What do I need to do after that? What is the expected behavior?

xobs, 2023-01-31:
Can you see if #311 fixes things?

bunnie, 2023-01-31:
The expected behavior is it /should/ just automatically run straight through and reformat the image, and then do a whole bunch of CI tests on the PDDB.

In hosted mode, the PDDB reformat is automatic -- you aren't actually running `xtask pddb-ci`, are you? I bet you're running some subset of the feature flags from `xtask`. All the flags need to be there for it to work, because the feature flags help to bypass questions like reformatting the PDDB and it also resizes the PDDB to make the test "harder" to pass.

xobs, 2023-01-31:
I'm running `cargo xtask pddb-ci` which builds things with these flags:

cargo build --release --package xous-ticktimer --package xous-log --package xous-names --package xous-susres --package graphics-server --package keyboard --package spinor --package llio --package com --package net --package dns --package gam --package ime-frontend --package ime-plugin-shell --package codec --package modals --package root-keys --package trng --package sha2 --package engine-25519 --package jtag --package status --package shellchat --package pddb --package usb-device-xous --features timestamp --features pddb/ci --features pddb/deterministic --features hosted

When it starts, this appears:

[image]

bunnie, 2023-01-31:
ah yes. That is part of the test, it exercises free space exhaustion. You just press the key and tell it you don't need to enumerate any more secret bases, and it will continue on.

Formatting the disk, on the other hand, would be a different question that would trigger if not all the arguments were present.

bunnie, 2023-01-31:
When it concludes you'll see this in the console:
```
INFO:pddb::backend::hw: derived bcrypt password in 5ms (services\pddb\src\backend\hw.rs:2593)
INFO:pddb::backend::basis: Basis Basis2 found and reconstructed (services\pddb\src\backend\basis.rs:1259)
INFO:pddb::tests: sanity check count: 50 (services\pddb\src\tests.rs:378)
INFO:pddb::tests: sanity check count: 50 (services\pddb\src\tests.rs:378)
INFO:pddb::tests: sanity check count: 50 (services\pddb\src\tests.rs:378)
INFO:pddb::tests: sanity check count: 65 (services\pddb\src\tests.rs:378)
INFO:pddb::tests: sanity check count: 65 (services\pddb\src\tests.rs:378)
INFO:pddb::tests: CI done (services\pddb\src\tests.rs:563)
```
Unfortunately, we can't quite hosted mode -- normally you'd shut down the program at that point, but it will idle there indefinitely.

bunnie, 2023-01-31:
It's working on my end now with #311 , I'll go ahead and close the issue.

closed 2023-01-31 by bunnie

xobs, 2023-01-31:
You can quit hosted mode, though it might not be what you want:

```patch
diff --git a/services/pddb/src/tests.rs b/services/pddb/src/tests.rs
index a5d0cae6..9dc6224a 100644
--- a/services/pddb/src/tests.rs
+++ b/services/pddb/src/tests.rs
@@ -562,6 +562,7 @@ pub(crate) fn ci_tests(pddb_os: &mut PddbOs) -> Result<()> {

         log::info!("CI done");

+        xous::rsyscall(xous::SysCall::Shutdown).unwrap();
         Ok(())
     }
 }
```

bunnie, 2023-01-31:
Well, it triggers a panic which causes the program to exit. So that's actually progress! That's helpful...I should try and find a way to integrate this into the regular CI pipeline then. thanks!

bunnie, 2023-01-31:
the problem was previously because the program wouldn't terminate, it was very hard to integrate into jenkins; at least it quits now, and I can tell jenkins to ignore the error code and then i can dig through the entrails that the test creates to determine pass or fail.