betrusted-io/xous-core #312: Suspend/resume regression with kernel changes

bunnie, 2023-01-31:
On resume, the device panics with this message:
```
PANIC in PID 5: panicked at 'assertion failed: arch::process::current_tid() == current.current_thread', kernel\src\services.rs:590:13
```
Any ideas what may be causing this?

xobs, 2023-01-31:
Yep, I see that in Renode. I added a consistency check, and that seems to be triggering on resume now.

xobs, 2023-01-31:
Something is setting `process.current_thread` to 0. I'm not sure what's causing that, because I never see anything assigning zero to it. The rest of the `process` is nonzero, so something is clearly subtly wrong.

I've worked around that in 1303e6416d5578531ee9230bfed90cae97304eab but it's still concerning.

bunnie, 2023-01-31:
3ee4bb2252d6de4165bfae3e855a0d4826c21c77 fixes an additional problem in the suspend engine itself.

The structure depended on blocking scalars not returning their quantum immediately. Now that this is fixed, the semantics of the call were broken.

This commit fixes that.

bunnie, 2023-01-31:
Looks like things are more stable now. I'm running a suspend/resume stress test that will take some time, but i'll go ahead and close this issue for now. thanks for the quick turn-around!

closed 2023-01-31 by bunnie