betrusted-io/xous-core #320: Add EH_FRAME and EH_FRAME_HDR flags to images

xobs, 2023-02-08:
It would be nice if we knew where EH_FRAME and EH_FRAME_HDR were so that we're ready to handle libunwind.

This needs to have the following happen:

* [ ] Define two new bits on `IniE` files to indicate a section is `EH_FRAME` or `EH_FRAME_HDR`: https://github.com/betrusted-io/xous-core/blob/main/docs/arguments.md#inie
* [ ] Add code to `create-image` to set those flags if `.eh_frame` or `.eh_frame_hdr` are encountered: https://github.com/betrusted-io/xous-core/blob/main/tools/src/elf.rs#L336-L344
* [ ] Modify the kernel so that it passes those addresses to the initial process: https://github.com/betrusted-io/xous-core/blob/main/kernel/src/services.rs#L368-L383
* [ ] Read these variables in the Rust `_start` routine and stash them in local variables: https://github.com/betrusted-io/rust/blob/28194d231b273388fce270ed5586c74738cba2e0/library/std/src/sys/xous/common.rs#L11-L14

bunnie, 2023-02-08:
Fantastic, thanks for the links. I'll poke at this once I've had some success (or I give up) at `xip` on the loader.

bunnie, 2023-02-08:
Note to self inside `create-image` the template looks a bit like this

```rust
if let Some(eh_frame_hdr) = elf_file.find_section_by_name(".eh_frame_hdr") {
            let addr = eh_frame_hdr.address().try_into()?;
            let len = (eh_frame_hdr.size()).try_into()?;
}
```
see https://github.com/xobs/horribly-patch-elf-files/blob/main/src/main.rs for more

bunnie, 2023-02-08:
Need to add something to the loader at or around `copy_processes()` that catches the new flags that are added, and stashes the offset and length of the section, so that it can be passed to the kernel later on. Probably shove them in the `InitialProcess` record within the `processes` field of `BootConfig`....

xobs, 2023-02-08:
The loader doesn't need to be modified at all. The arguments block is already passed to the kernel, so the kernel can take care of reading those values when it sets up processes in https://github.com/betrusted-io/rust/blob/28194d231b273388fce270ed5586c74738cba2e0/library/std/src/sys/xous/common.rs#L11-L14

bunnie, 2023-02-15:
@xobs I have a PR open both in this repo and in `rust` to implement this.

I wasn't quite sure where to stick the frame header stuff, so I just created a `static AtomicUsize`. I had a hack to confirm things worked by passing them as arguments to a `std` call that wasn't using them and printing it out; obviously that is not committed in the final version, but it looks like the numbers are correct.

If the PRs look good to you please merge & close the issue. thanks!

commit 08ae66a2 (2023-02-15): fix: #320

this commit adds the facilities to extract eh_frame[_header]
sections and pass it to each process.

A bump is required to `std` to extract the passed parameters.
commit 13b614a2 (2023-02-15): Merge pull request #322 from betrusted-io/eh_header

fix: #320 adds the eh_frame tags and passes them to each process
closed 2023-02-15 by xobs