betrusted-io/xous-core #326: make EC updates more foolproof

bunnie, 2023-02-28:
- [ ] add some UX lag before starting the actual update process. This will give some warning to users who reflexively reset their devices after doing an update, so that an EC update is not accidentally disrupted
- [ ] add a link test before doing the EC update. Don't do the update if the link seems unstable.

nworbnhoj, 2023-02-28:
I undertook an update with a new USB cable and did not physically touch Precursor during the update. The Update appeared to complete successfully and the Precursor appeared to reboot.

```
Resuming CPU.
Resetting SOC...

Update finished!
=== QR-code redacted ===
Visit the QR code above to help locate the hole, or go to https://ci.betrusted.io/i/reset.jpg.
You *must* reboot by inserting a paperclip in the hole in the lower right hand side, then follow the on-device instructions.
```

At this point I physically picked up the Precurrsor and paperclip reset with the USB cable still attached. The Precursor reset and became stuck on `Starting up...` and a modal `EC Autoupdate Erasing (gateware)`. 

I proceeded with the paperclip reset (despite the Precursor apparently resetting automatically), because of the instruction `You *must* reboot by inserting a paperclip....` and that I had read on the forum that a reset was mandatory for the EC to take properly.

Perhaps the instructions would be better as:
```
Resuming CPU.
Resetting SOC...

Update finished!

Precursor should reset automatically at this point - and you should follow the on-device instructions.

If the Precursor does not reset automatically then you *must* reboot by inserting a paperclip in the hole in the lower right hand side - and then follow the on-device instructions.

Need help locating the reset hole? Go to https://ci.betrusted.io/i/reset.jpg.

QR-code

```

bunnie, 2023-03-08:
Implemented the fixes + updated the text. thanks for the feedback @nworbnhoj !

commit fb3dabbd (2023-03-08): fix #326

- add some ux lag
- checks link integrity

previous commit also revises message on precursorupdater
per @nworbnhoj's comments.
closed 2023-03-08 by bunnie