betrusted-io/xous-core #334: Don't try to save all '0' U2F Keys

kop316, 2023-03-23:
In Gitlab, you can register a device. But if you forget to actually type a name, the Precursor will have a U2F key for a username but Gitlab does not. In this case, Gitlab spams an all '0's app and you can accidentally save it, which causes issues with vault complaining about a U2F key being invalid.

It would be nice if the vault app could ignore the all '0's key

bunnie, 2023-03-24:
I *think* what you're referring to is the app_id. So what you're saying is, if the app_id is all 0, don't try to prompt for a local name for the app_id?

kop316, 2023-03-24:
ahh, yes that is what I mean, sorry! yes if the app_id is all zeros.

bunnie, 2023-03-24:
So in this case, an app_id of all 0's may show up and request authorization, but you'll never be prompted to give it a name. Is that the desired behavior?

kop316, 2023-03-24:
That would be yes! In the current state, I was prompted to give it a name, and it saved (or I had no option to not save). But then it leads to the error here: 

https://bot-matrix.ems.host/_matrix/media/r0/download/bot-matrix.ems.host/c48a7d1362587acbadd8d964a3a1b29155ef0a19

To which I can only delete with `pddb keydelete fido.u2fapps:0000....000` (which isn't easy to do...) or delete all of the u2f keys with `pddb dictdelete fido.u2fapps`, but then I have to re-add everything back

bunnie, 2023-03-24:
I don't have away to test it, but i just pushed a commit that implements the behavior I think you described.

kop316, 2023-03-24:
I have a way to test (I at least know how I got there in the first place), will let you know!

kop316, 2023-03-24:
I confirmed it does not save. Thanks!

closed 2023-03-24 by kop316