betrusted-io/xous-core #338: Feature request: Precursor as a hardware entropy source

markhuge, 2023-03-26:
Really handy for things like generating keys. I believe [rngd](https://docs.kernel.org/admin-guide/hw_random.html) reads a /dev/urandom style stream over serial

Notes:
- rng-tools moved from [sourceforge](https://sourceforge.net/projects/gkernel/) to [github](https://github.com/nhorman/rng-tools).
-  [Prior art](https://www.chucknemeth.com/usb-devices/onerng) with [OneRNG](https://onerng.info/)

bunnie, 2023-04-16:
#365 I believe implements the basics for this feature.

running `usb trng` while the device is plugged into a host should allow it to function as an entropy source. 

I *think* it emulates the interface used by OneRNG, but I'd love it if someone who actually uses these dongles could verify that.

bunnie, 2023-05-16:
Since this is implemented, but nobody has tested it, I'm going to close this for now, and when someone finds a problem with the implementation they should open a new issue with steps to reproduce and I can address it.

closed 2023-05-16 by bunnie

markhuge, 2023-05-16:
@bunnie I've been testing. Have not been able to get rngd to recognize the precursor as an entropy source just yet, but assuming PEBKAC at the moment

bunnie, 2023-05-17:
Alright. I am suspecting there might be some out of band protocol that's expected by the TRNG daemons to handshake but I can't find any specs on it. If you can send details on how to reproduce an issue (assuming a clean system) that would be helpful.

Part of the problem is I don't have one of these HW dongles myself so I can't just hook up a protocol analyzer to see what the setup handshake (if any) looks like, but as far as I can see there shouldn't be any, it just wants to pull data from the serial port...

markhuge, 2023-05-21:
Quick update. I used this as an excuse to upgrade my oscilloscope. Sourcing some known working dongles. Will put some time into this once all the hardware arrives. Will report back.

bunnie, 2023-05-28:
Update on my side:

- Requires use of the latest rngd tools. The version 2.2 packaged into e.g. ubuntu/debian will declare an RNG as unfit for use if it can't respond fast enough on the first try to pull for data (considers `EAGAIN` to be a hard failure). later versions do not fail in this way.
- `./rngd -f -x rdrand -r /dev/ttyACM0 -t` + protocol analyzer indicates that entropy is flowing, and also, unplugging the entropy source during this test causes things to fail.

Log:
```
./rngd -f -x rdrand -r /dev/ttyACM0 -t
Disabling 2: Intel RDRAND Instruction RNG (rdrand)
Initializing available sources
[hwrng ]: Initialized
[pkcs11]: No PKCS11 endpoint /usr/lib64/opensc-pkcs11.so found
[pkcs11]: Install opensc/opensc-pkcs11/etc if you would like to gather smartcard entropy
[pkcs11]: Initialization Failed
[rtlsdr]: Initialization Failed
Entering test mode...no entropy will be delivered to the kernel
Entropy gathered: 2.000000e+04 bits
Entropy gathered: 4.000000e+04 bits

...

Entropy gathered: 2.000000e+04 bits
Entropy gathered: 2.000000e+04 bits

<< device is unplugged >>

Too Slow: Disabling Hardware RNG Device
No entropy sources working, exiting rngd

Entropy rate:  20.69 Kbits/sec averaged over 26 iterations for  39.65 seconds
``` 

So I think it's working, but, you do need to install a more recent version of rng-tools. The version used for my testing is fetched from

https://github.com/nhorman/rng-tools