betrusted-io/xous-core #341: Xous does not support open WiFi networks

kop316, 2023-03-27:
Right now, Xous does not support open WiFi Networks. It would be nice if it did.

bunnie, 2023-07-15:
This should be fixed in https://github.com/betrusted-io/xous-core/commit/1434764f086a7958e7cd2c2472e7cc8f52f95b06, but you also need to update the EC.

I recommend running the `precursorupdater -b` to pull in the full set of artifacts automatically.

If someone could please check that this works for them, that would be great. The full extent of my testing was "did it connect, did it ping"? yes and yes, SHIP IT.

kop316, 2023-07-17:
ok, I updated (and the EC was updated too). I also saw the screen to delete the PW placeholder. However, I just see it stay at `Not Connected`. I don't see any debugging, so I am not sure what useful things I can show you.

bunnie, 2023-07-17:
Can you share what `Network Status` shows in the wifi preferences submenu?

kop316, 2023-07-17:
> Can you share what `Network Status` shows in the wifi preferences submenu?

Heh, ok well I deleted the network then tried again, it works! Maybe I had it before this worked? But anyways, this can be closed.

closed 2023-07-17 by kop316

bunnie, 2023-07-17:
Cool. it's also possible that...it just failed to DHCP the first time around, and it takes a long time for it to try again. Who knows...wireless. it's magic!